eclipse-threadx/usbx #5: Where are the host controller drivers?

dreiss, 2020-12-19:
The documentation overview ( https://docs.microsoft.com/en-us/azure/rtos/usbx/overview-usbx ) says "Azure RTOS USBX supports proprietary discrete USB host controllers from Atmel, Microchip, Philips, Renesas, ST, TI, and other vendors".  However, under `common/usbx_host_controllers`, I only see drivers for OHCI and EHCI.  Are there drivers (and, ideally, sample code) for using USBX Host on STM32F4 or Microchip SAMD51?

LP-33, 2020-12-22:
Same question here, as we would like to use USBX in a SAM D21 device.

Also, in general, who is responsible for the ports of the Azure-RTOS components?
Having the components/framework source code is nice, but if we have to write the BSP/glue code there is no way to reduce the time to market when using a pre-licensed hardware.

goldscott, 2020-12-22:
Hi all, we have a handful of host controller drivers available - I'm not sure why we haven't posted them on github yet. If you look at the example projects in the [samples repo](https://github.com/azure-rtos/samples) you can find drivers for ST and SAM E54. I'll find out more about other host controllers.

dreiss, 2020-12-29:
Thanks for the pointer!  In the "stm32f746g-disco" sample, I found the source file `./stm32cubeide/sample_usbx_host_mass_storage/usbx_host_controllers/ux_hcd_stm32_initialize_fscore.c`, which appears to be what I would need to run the OTG_FS peripheral in the STM32F411 in host mode (though I haven't tried it out yet).  It would be awesome to see this file added to the main USBX repository, and even more awesome to see a demo of using STM32F411 in host mode, since the middleware supplied by the chip manufacturer does not support USB hubs.

goldscott, 2020-12-31:
There is a demo of host mode for 746 - see sample_usbx_host_mass_storage. The 411 device is probably pretty similar.

yuxin-azrtos, 2021-01-28:
Close this issue.  Feel free to reopen it if you have additional feedback.

closed 2021-01-28 by yuxin-azrtos