eclipse-threadx/usbx #6: USBX Host : Multiple CDC device instances

luis-casado, 2021-01-29:
Hello,

Is it possible to connect more than one CDC device to USBX host stack? Or two independent USB CDC devices or a single device with two CDC interfaces. First one is detected correctly, but not sure how to implement the second interface that has been enumerated and properly configured.

xiaocq2001, 2021-02-01:
Yes. It's possible to connect more than one CDC device, or composite device with two associated CDC interfaces (CDC interfaces must be grouped by IAD, that is, IAD+CDC-ACM+CDC-DATA for each CDC function).

Only one CDC-ACM class stack is registered for multiple CDC functions.
For each CDC function connected, an instance is created by USBX host stack, and inform application by `ux_system_host_change_function` registered on host stack initialization.

Also the instance can be get from the CDC-ACM class:

```
/* Get CDC-ACM class (if registered with _ux_system_host_class_cdc_acm_name).  */
ux_host_stack_class_get(_ux_system_host_class_cdc_acm_name, &cdc_class);

/* Get first activated CDC-ACM function instance.  */
ux_host_stack_class_instance_get(cdc_class, 0, (void **) &cdc_acm0);

/* Get second activated CDC-ACM function instance.  */
ux_host_stack_class_instance_get(cdc_class, 1, (void **) &cdc_acm1);
```

luis-casado, 2021-02-01:
Thank you very much! I will tested it.

yuxin-azrtos, 2021-02-05:
@luis-casado Did you get a chance to test it? Let us know. Thanks!

luis-casado, 2021-02-08:
Hello @yuxin-azrtos ,

Yes, I have tested it with no success.  I have got two different CDC devices working separately and working through USB hub in order to prepare the test of both  connected to USB HUB. 
I have added the code to work with the second instance, but even with only one CDC device connected,:

ux_host_stack_class_instance_get(cdc_class, 1, (void **) &cdc_acm1);

is returning UX_SUCCESS. 

The code should wait for the second instance connected and to be alive, but with only one device CDC connected, this code is not waiting for the second instance:

/* We get the second instance of the CDC device */
     do  
      {
          status =  ux_host_stack_class_instance_get(cdc_host_class,, 1, (void **) &cdc_acm1);
          tx_thread_sleep(10);
      } while (status != UX_SUCCESS);

         /* We still need to wait for the CDC1 status to be live */
        while (cdc_acm1 -> ux_host_class_cdc_acm_state != UX_HOST_CLASS_INSTANCE_LIVE)
        tx_thread_sleep(10);

luis-casado, 2021-02-08:
Hello @yuxin-azrtos ,

It looks like status = ux_host_stack_class_instance_get(cdc_host_class,, 1, (void **) &cdc_acm1);
is returning the same instance of cdc. I can communicate with CDC device using cdc_acm0 and cdc_acm1.

Any thought?

Luis

luis-casado, 2021-02-08:
In additional test,  with the two cdc devices connected, I receive and send data with both instances, acm0 and acm1 and pointing both instances to the same USB CDC device.

xiaocq2001, 2021-02-09:
Host CDC ACM creates two instances for two interfaces inside CDC function (Normal sequence is CDC interface and CDC data interface).
```
    /* Find the main cdc_acm container */
    status = ux_host_stack_class_get(_ux_system_host_class_cdc_acm_name, &class);
    if (status != UX_SUCCESS)
        /* Do not break. */
        return 0;

    /* Find class instance. */
    status = ux_host_stack_class_instance_get(class, 0, (void **) &cdc_acm_host_control);
    if (status != UX_SUCCESS)
        /* Do not break. */
        return 0;

    if (cdc_acm_host_control->ux_host_class_cdc_acm_state != UX_HOST_CLASS_INSTANCE_LIVE)
        /* Do not break. */
        return 0;

    /* Find class instance. */
    status = ux_host_stack_class_instance_get(class, 1, (void **) &cdc_acm_host_data);
    if (status != UX_SUCCESS)
        /* Do not break. */
        return 0;

    if (cdc_acm_host_control->ux_host_class_cdc_acm_state != UX_HOST_CLASS_INSTANCE_LIVE)
        /* Do not break. */
        return 0;
```
Also when you get CDC instance you can check interface class to see if it's control or data: `cdc_inst -> ux_host_class_cdc_acm_interface -> ux_interface_descriptor.bInterfaceClass == UX_HOST_CLASS_CDC_CONTROL_CLASS`

So two CDC device have 4 CDC instance ready to operate, 2 for control and 2 for data, you can use data instances for data transfer.

luis-casado, 2021-02-09:
Hello @xiaocq2001 ,

Great, it is working now. I was wrong, I was using the instance 0 and managing manually the pointer to NEXT_INSTANCE in a wrong way. Using index works as expected. Thank you very much!
I see the support for Prolific USB2SER class is included in the stack. Do you have support for FTDI and/or Silicon Labs USB2SER devices?

Luis

yuxin-azrtos, 2021-04-12:
Closing the issue as it is resolved.

closed 2021-04-12 by yuxin-azrtos

BarryAndrews, 2023-02-01:
Is there any example code of the creation of multiple CDC instances and the handling of reading from these?
I have same requirement, but cannot find any examples where the HOST CDC creates multiple instances of CDC to allow for the multiple endpoints where device has more than one CDC channel.

xiaocq2001, 2023-02-01:
When a device with multiple CDC channel connected and enumerated, if CDC class is registered, the instances are created on enumeration and the instances linked to CDC DATA interfaces are ready for reading/writing.

So what you do in application is to register the class, check activated CDC DATA instances and use them to read/write.

BTW, multiple CDC channel consumes more endpoints, please confirm your hardware and HCD driver support the number of endpoints. For each CDC-ACM 3 endpoints are needed. E.g., for device with two CDC-ECM channel needs 1 (EP0) + 3 * 2 = 7 endpoints.

BarryAndrews, 2023-02-01:
Thanks for the quick response.
The example I have is from the ST version of USBX and in this case they only create a single instance for any CDC device connection.

In the host callback
`UINT ux_host_event_callback(ULONG event, UX_HOST_CLASS *Current_class, VOID *Current_instance)
`
The instance isn't added to an existing list, it is just set

`			  USBH_UsrLog("USB cdc_acm device insertion.");
			  if (ux_device_class == Current_class)
			  {
				usb_ptr->app_cdc_acm = Current_instance;
				usb_ptr->ux_dev_info.Device_Type = CDC_ACM_Device;

				/* Check if this is a data interface */
				if (usb_ptr->app_cdc_acm->ux_host_class_cdc_acm_interface->ux_interface_descriptor.bInterfaceClass == UX_HOST_CLASS_CDC_DATA_CLASS)
				{
				  tx_queue_send(&usb_ptr->ux_app_MsgQueue, &usb_ptr->ux_dev_info, TX_NO_WAIT);
				}
			  }
			  else
			  {
				  USBH_UsrLog("Not a CDC Class Device");
				  USBH_UsrLog("Class Name: %s", Current_class->ux_host_class_name);
				  USBH_UsrLog("Expected Class Name: %s", _ux_system_host_class_cdc_acm_name);
			  }
`

The instance itself is allocated from memory in the "_ux_host_class_cdc_acm_activate" function and doesn't allow for any more than single CDC device with single pair of data endpoints

The usb_ptr->app_cdc_acm is a UX_HOST_CLASS_CDC_ACM *, so could also contain an array.  I was hoping for an example where this is used to hold an array rather than a single instance so that I can see which functions I should be using and how I cycle through all the read endpoints.

Thanks for the info regards endpoints, I was aware of this and have sufficient but it's always good to remember.

xiaocq2001, 2023-02-02:
I just checked the example generated from STMCube.

The USBX side library should be the same as 6.1.9 release here.

BarryAndrews, 2023-02-02:
I am using STM32 MX Cube version 6.7.0
I previously generated code based on the Host_CDC_ACM examples from ST.
To check again, I downloaded Ux_Host_CDC_ACM for the H7 variant of Nucleo-144 using MX CUBE
The ux_host_event_callback is exactly the same as my current example, it just saves the pointer to the Current_instance to the  app_cdc_acm, there is no linked list and only the one activation.

The _ux_host_class_cdc_acm_activate function is called once  in _ux_host_class_cdc_acm_entry and not called any other times.
I have a single CDC device which comes from the one interface but has 1 control channel and 4 separate data channels (of which I wish to only use two), ultimately I want two of these devices on separate HCD, giving me 2 control channels, 4 data channels.  
I am able the enumerate and correctly locate the endpoints for one of my Vendor devices, but they are then all contained within one vendor class which has the interface, control channel and two data channels.  

Currently the stack adds this vendor class to the stack as one CDC device and hence only reads the first data channel. I can split this one class into two so both are added to the stack but I want an example to make sure I do this in the correct location in the code and execute the transfer requests correctly.  I am not sure if these need to be sequentially or can be done in parallel.  Your earlier information suggest parallel, which I assume means I need separate callback functions for each transfer on a data channel.

Your indication is that the example already shows multiple classes being registered and called back.
Which example are you referring to so that I can install it?

xiaocq2001, 2023-02-02:
There is no existing example to save CDC instances to array.

From your description, it seems the device that has 4 data channels is not what the stack expected.

It's better to get a configuration descriptor reported from your device to analyze. If you can provide the descriptors we can analyze together.

Without descriptor, to analyze in general,

`_ux_host_class_cdc_acm_entry` is invoked with command `UX_HOST_CLASS_COMMAND_ACTIVATE` to do activate function, so it's expected that `ux_host_event_callback` is invoked multiple times (you can set breakpoint in it to check) if standard CDC-ACM device is connected.

For standard CDC-ACM each channel includes two interfaces as control and data, if there are multiple channels there should be multiple pair of control and data interfaces.

If your device has only one control interface but contains multiple of data interfaces, maybe it needs additional driver to work on PC (windows or linux, etc.), in such case it's not quite compatible with current stack.

BTW, the interface I mentioned is what in USB reported descriptor, but not physical port or connectors of your device hardware.

BarryAndrews, 2023-02-02:
Thanks for the response.
It seems your implementation is for Windows, mine is for STM32H7 processor on its own PCB running the Azure ThreadX  OS.
I am enumerating the connection without any issues and getting all of the end points.

I am storing these end points into a VENDOR class based on the CDC_ACM struct but having two sets of in end points, two sets of out end points and two interrupt endpoints

`typedef struct UX_HOST_CLASS_CDC_ACM_STRUCT
{
    struct UX_HOST_CLASS_CDC_ACM_STRUCT  
                   *ux_host_class_cdc_acm_next_instance;
    UX_HOST_CLASS  *ux_host_class_cdc_acm_class;
    UX_DEVICE      *ux_host_class_cdc_acm_device;
    UX_ENDPOINT    *ux_host_class_cdc_acm_bulk_in_endpoint;
    UX_ENDPOINT    *ux_host_class_cdc_acm_bulk_out_endpoint;
    UX_ENDPOINT    *ux_host_class_cdc_acm_interrupt_endpoint;
    UX_INTERFACE   *ux_host_class_cdc_acm_interface;
    UINT           ux_host_class_cdc_acm_instance_status;
    UINT           ux_host_class_cdc_acm_state;
    ULONG          ux_host_class_cdc_acm_notification_count;
    ULONG          ux_host_class_cdc_acm_capabilities;
    ULONG          ux_host_class_cdc_acm_device_state;
    struct UX_HOST_CLASS_CDC_ACM_RECEPTION_STRUCT  
                   *ux_host_class_cdc_acm_reception;
    
    VOID           (*ux_host_class_cdc_acm_device_status_change_callback)(struct UX_HOST_CLASS_CDC_ACM_STRUCT *cdc_acm, 
                                                                ULONG  notification_type, ULONG notification_value);
    UX_SEMAPHORE   ux_host_class_cdc_acm_semaphore;
} UX_HOST_CLASS_CDC_ACM;
`
All the functions for enumeration need to have this full structure with all the endpoints.
I might try adding a second set of the reception struct for my second in channel and duplicate the receive code, one for each reception struct.