eclipse-threadx/usbx #9: Azure RTOS USBX audio class 2.0

ayedm1, 2021-03-22:
Hi All,

I’m Currently working on an USBx audio application with an stm32H743i. I have some point for support & enhancement .

I found a mistake when registering audio class in MS website https://docs.microsoft.com/fr-fr/azure/rtos/usbx/usbx-device-stack-supplemental-2#usb-device-audio-class

[image]

Support. 

**issue**
I can’t enumerate my audio2.0 device. 
This is a trace of enumerating phase with Lecroy analyser. 

[image]

I’ have as reference a trace of enumeration of an USB-Audio device 2.0.

[image]

**Enhance**
In “ux_device_class_audio20_control_process.c” file it’s possible to set WRES in applicative side by user when host request FU_VOLUME_CONTROL?

[image]

For feature unit control selectors macros in  “ux_device_class_audio20.h” it is possible to change it like Audio device Class Specification.

For Example change UX_DEVICE_CLASS_AUDIO20_FU_MUTE_CONTROL 0x01 to 0x03

[image]

Screenshot from audio20 spec sheet 59

[image]

Regards,
Mohamed

xiaocq2001, 2021-03-25:
1. About document, thanks for reporting this, we will update that.
2. About enumerate, it seems host stopped the process (since there seems no error in trace log), maybe their is some issue in descriptors or request returns that host is not compatible.
Following is example descriptor, initializing and request handling for USB Audio 2.0 speaker:
```
#define D3(d) ((UCHAR)((d) >> 24))
#define D2(d) ((UCHAR)((d) >> 16))
#define D1(d) ((UCHAR)((d) >> 8))
#define D0(d) ((UCHAR)((d) >> 0))
```
```
/* --------------------------------------- Device Descriptor  */
/* 0  bLength, bDescriptorType                                */ 18,   0x01,
/* 2  bcdUSB                                                  */ D0(0x200),D1(0x200),
/* 4  bDeviceClass, bDeviceSubClass, bDeviceProtocol          */ 0x00, 0x00, 0x00,
/* 7  bMaxPacketSize0                                         */ 0x08,
/* 8  idVendor, idProduct                                     */ 0x84, 0x84, 0x03, 0x00,
/* 12 bcdDevice                                               */ D0(0x200),D1(0x200),
/* 14 iManufacturer, iProduct, iSerialNumber                  */ 0,    0,    0,
/* 17 bNumConfigurations                                      */ 1,

/* ----------------------------- Device Qualifier Descriptor  */
/* 0 bLength, bDescriptorType                                 */ 10,                 0x06,
/* 2 bcdUSB                                                   */ D0(0x200),D1(0x200),
/* 4 bDeviceClass, bDeviceSubClass, bDeviceProtocol           */ 0x00,               0x00, 0x00,
/* 7 bMaxPacketSize0                                          */ 8,
/* 8 bNumConfigurations                                       */ 1,
/* 9 bReserved                                                */ 0,

/* -------------------------------- Configuration Descriptor  *//* 9+8+73+55=145 */
/* 0 bLength, bDescriptorType                                 */ 9,    0x02,
/* 2 wTotalLength                                             */ D0(145),D1(145),
/* 4 bNumInterfaces, bConfigurationValue                      */ 2,    1,
/* 6 iConfiguration                                           */ 0,
/* 7 bmAttributes, bMaxPower                                  */ 0x80, 50,

/* ------------------------ Interface Association Descriptor  */
/* 0 bLength, bDescriptorType                                 */ 8,    0x0B,
/* 2 bFirstInterface, bInterfaceCount                         */ 0,    2,
/* 4 bFunctionClass, bFunctionSubClass, bFunctionProtocol     */ 0x01, 0x00, 0x20,
/* 7 iFunction                                                */ 0,

/* ------------------------------------ Interface Descriptor  *//* 0 Control (9+64=73) */
/* 0 bLength, bDescriptorType                                 */ 9,    0x04,
/* 2 bInterfaceNumber, bAlternateSetting                      */ 0,    0,
/* 4 bNumEndpoints                                            */ 0,
/* 5 bInterfaceClass, bInterfaceSubClass, bInterfaceProtocol  */ 0x01, 0x01, 0x20,
/* 8 iInterface                                               */ 0,
/* ---------------- Audio 2.0 AC Interface Header Descriptor *//* (9+8+17+18+12=64) */
/* 0 bLength, bDescriptorType, bDescriptorSubtype            */ 9,                   0x24, 0x01,
/* 3 bcdADC, bCategory                                       */ D0(0x200),D1(0x200), 0x08,
/* 6 wTotalLength                                            */ D0(64),D1(64),
/* 8 bmControls                                              */ 0x00,
/* -------------------- Audio 2.0 AC Clock Source Descriptor */
/* 0 bLength, bDescriptorType, bDescriptorSubtype            */ 8,    0x24, 0x0A,
/* 3 bClockID, bmAttributes, bmControls                      */ 0x10, 0x05, 0x01,
/* 6 bAssocTerminal, iClockSource                            */ 0x00, 0,
/* ------------------- Audio 2.0 AC Input Terminal Descriptor */
/* 0  bLength, bDescriptorType, bDescriptorSubtype            */ 17,   0x24,                   0x02,
/* 3  bTerminalID, wTerminalType                              */ 0x04, D0(0x0101),D1(0x0101),
/* 6  bAssocTerminal, bCSourceID                              */ 0x00, 0x10,
/* 8  bNrChannels, bmChannelConfig                            */ UX_DEMO_NB_CHANNEL, D0(0),D1(0),D2(0),D3(0),
/* 13 iChannelNames, bmControls, iTerminal                    */ 0,    D0(0),D1(0),            0,
/* --------------------- Audio 2.0 AC Feature Unit Descriptor */
/* 0  bLength, bDescriptorType, bDescriptorSubtype            */ 18,   0x24, 0x06,
/* 3  bUnitID, bSourceID                                      */ 0x05, 0x04,
/* 5  bmaControls(0), bmaControls(...) ...                    */ D0(0xF),D1(0xF),D2(0xF),D3(0xF), D0(0),D1(0),D2(0),D3(0), D0(0),D1(0),D2(0),D3(0),
/* .  iFeature                                                */ 0,
/* ------------------ Audio 2.0 AC Output Terminal Descriptor */
/* 0  bLength, bDescriptorType, bDescriptorSubtype            */ 12,          0x24,                 0x03,
/* 3  bTerminalID, wTerminalType                              */ 0x06,        D0(0x0301),D1(0x0301),
[... 71 lines cut ...]

    /* Handle request and update control values.
       Note here only mute and volume for master channel is supported.
     */
    status = ux_device_class_audio20_control_process(audio, transfer, &group);
    if (status == UX_SUCCESS)
    {
        /* Request handled, check changes */
        switch(audio_control[0].ux_device_class_audio20_control_changed)
        {
        case UX_DEVICE_CLASS_AUDIO20_CONTROL_MUTE_CHANGED:
[... cut ...]
    return(status);
}
```
3. Thanks for the suggestion to support wRES, we will add that in our basic request handling function. Note that since their are so many audio entities we are not able to do support for all of them, you can do request handling in `demo_audio20_request_process` anyway, to have specific things supported.
4. UX_DEVICE_CLASS_AUDIO20_FU_MUTE_CONTROL is actually for **Control Selectors** in Table A-23, Section A.17.7 but **NOT bmaControls bits**, the bmaControls bits are not defined.

ayed-m, 2021-03-25:
Hi CQ Xiao, 

Thank you for this feedback. Can you please share with me this macro value to test if i have the same behaviour :).
UX_DEMO_SAMPLING_FREQUENCY, 
UX_DEMO_VOLUME_MIN, 
UX_DEMO_VOLUME_MAX,
UX_DEMO_VOLUME_DEFAULT, 
UX_DEMO_FRAME_BUFFER_NB,
 UX_DEMO_MAX_FRAME_SIZE

Regards
Mohamed

ayedm1, 2021-03-25:
Hi CQ Xiao,

Thank you for this feedback. Can you please share with me this macro value to test if i have the same Behaviour :).
UX_DEMO_SAMPLING_FREQUENCY,
UX_DEMO_VOLUME_MIN,
UX_DEMO_VOLUME_MAX,
UX_DEMO_VOLUME_DEFAULT,
UX_DEMO_FRAME_BUFFER_NB,
UX_DEMO_MAX_FRAME_SIZE,

Regards
Mohamed

xiaocq2001, 2021-03-26:
```
#define UX_DEMO_SAMPLING_FREQUENCY          AUDIO_FREQUENCY_44K /* from ST BSP */
#define UX_DEMO_NB_CHANNEL                  2
#define UX_DEMO_SAMPLE_SIZE                 2
#define UX_DEMO_SUBFRAME_SIZE               (UX_DEMO_SAMPLE_SIZE * UX_DEMO_NB_CHANNEL)
#define UX_DEMO_FRAME_NB_SUBS               (UX_DEMO_SAMPLING_FREQUENCY / 1000)
#define UX_DEMO_FRAME_SIZE                  (UX_DEMO_FRAME_NB_SUBS * UX_DEMO_SUBFRAME_SIZE)

#define UX_DEMO_MAX_FRAME_SIZE              256
#define UX_DEMO_FRAME_BUFFER_NB             2

#define UX_DEMO_VOLUME_MIN                  -32767
#define UX_DEMO_VOLUME_MAX                  32767
#define UX_DEMO_VOLUME_RES                  1
#define UX_DEMO_VOLUME_DEFAULT              0
```

yuxin-azrtos, 2021-04-12:
@ayedm1 any updates?

ayedm1, 2021-04-14:
Hi @yuxin-azrtos, for me enumeration is OK. I will continue to dev my app now.

yuxin-azrtos, 2021-04-14:
@ayedm1  Thanks! Let us know if you have additional questions.

closed 2021-04-14 by yuxin-azrtos