eclipse-threadx/usbx #10: Where are hardware-specific drivers, such as those for STM32?

trinhnhatnam, 2021-04-07:
Hello,

I got the drivers from the samples repo, but shouldn't they be included in this repo?

Thanks!

yuxin-azrtos, 2021-04-07:
ST is in the process of integrating USBX with their MCU offering.  Please contact ST for availability and schedule.

trinhnhatnam, 2021-04-07:
For example, in the samples repo, I can see files such as _ux_hcd_stm32_asynch_schedule.c with the following comment:

~~~
/**************************************************************************/ 
/*                                                                        */ 
/*  FUNCTION                                               RELEASE        */ 
/*                                                                        */ 
/*    _ux_hcd_stm32_asynch_schedule                       PORTABLE C      */ 
/*                                                           6.1          */
/*  AUTHOR                                                                */
/*                                                                        */
/*    Chaoqiong Xiao, Microsoft Corporation                               */
/*                                                                        */
/*  DESCRIPTION                                                           */
/*                                                                        */ 
/*    This function schedules a new transfer from the control/bulk lists. */ 
/*                                                                        */ 
/*  INPUT                                                                 */ 
/*                                                                        */ 
/*    hcd_stm32                           Pointer to STM32 controller     */ 
/*                                                                        */ 
/*  OUTPUT                                                                */ 
/*                                                                        */ 
/*    TRUE or FALSE                                                       */ 
/*                                                                        */ 
/*  CALLS                                                                 */ 
/*                                                                        */ 
/*                                                                        */ 
/*  CALLED BY                                                             */ 
/*                                                                        */ 
/*    STM32 Controller Driver                                             */
/*                                                                        */ 
/*  RELEASE HISTORY                                                       */ 
/*                                                                        */ 
/*    DATE              NAME                      DESCRIPTION             */ 
/*                                                                        */ 
/*  05-19-2020     Chaoqiong Xiao           Initial Version 6.0           */
/*  09-30-2020     Chaoqiong Xiao           Modified comment(s),          */
/*                                            resulting in version 6.1    */
/*                                                                        */
/**************************************************************************/
~~~

So those files are from Microsoft, not ST. My question is what aren't they included in this repo?

yuxin-azrtos, 2021-04-12:
Closing this issue.  Feel free to reach out if you have additional questions.

closed 2021-04-12 by yuxin-azrtos

trinhnhatnam, 2021-04-13:
I'm sorry, but my question hasn't been answered. Why aren't hardware-specific files, such as the _ux_hcd_stm32_asynch_schedule.c file I mentioned, included in this repo? It's a fair question.

The USB stack repo from Micrium, for example, includes hardware-specific files.

yuxin-azrtos, 2021-04-13:
@trinhnhatnam, ST already added USBX support (with proper drivers) for their STM32H7 family MCUs: https://www.st.com/en/embedded-software/x-cube-azrtos-h7.html

And ST is in the process of expanding such support into more MCUs (such as their L4 family, F7 and F4 family MCUs).    

These drivers are maintained by ST, therefore Microsoft does not re-distribute the drivers in Azure RTOS repo.

trinhnhatnam, 2021-04-13:
Can you forget about what ST is doing for a moment?

I'm asking about the hardware-specific files for USBX that I see in the "samples" repo. Those are clearly maintained by Microsoft, with proper comments and versioning. Why aren't they in this repo?

goldscott, 2021-04-13:
@trinhnhatnam Let me try to clear things up here. Microsoft doesn't really want to write and maintain drivers and BSPs for hundreds of different devices, which is why we've partnered with a handful of popular semiconductor manufacturers. Microsoft wrote drivers to get those samples working on real hardware. We got the samples up on github for people to start using right away while ST, Renesas, NXP, Microchip, etc. work on integrating Azure RTOS into their IDEs. The semis are going to be providing the drivers for their hardware. They want to be the one-stop-shop for developers using their devices (e.g. you download stm32cube IDE and everything is in there - Azure RTOS, drivers, sample code, etc). They want to "curate" the "developer experience." 

You're not the only one that has asked about USBX drivers. I'm an Azure RTOS developer and I've asked if we're going to put drivers up on github. We have a lot of old drivers (from the Express Logic days), but that's just it - they're for older and obsolete devices. It's actually nice that the semis want to write and maintain the drivers, because they know their hardware much better than we do, so they can write drivers that take full advantage of all the hardware capabilities.

trinhnhatnam, 2021-04-13:
Why not put all those "old" and "obsolete" drivers up anyway? They're already available. They can help someone, like me, who tries to switch to AzureRTOS stuff. My devices are not new to wait for manufacturers to write "good" drivers for. For example, an ST device of mine is not an H7; I don't want to wait for ST to write something for it. I can tell you that your "obsolete" drivers really help me.

And if you worry about the obsoleteness, then put up a warning directing users to  semiconductor manufacturers for possible newer and better drivers.

And even if the semiconductor manufacturers are going to write all drivers from now on, I think Microsoft should make an effort to make this an independent product by trying to bring those drivers over to this repo. One obvious reason, in ST case, is that not everyone likes stm32cubeide, they may prefer IAR for example.

This is my humble opinion. You don't have to reply. Thanks @yuxin-azrtos and @goldscott.

goldscott, 2021-04-13:
@trinhnhatnam - I agree with you. The old drivers are helpful. But it's not my decision to put them up.