eclipse-threadx/usbx #12: usbx_device_controllers

geek-slt, 2021-04-14:
Hello, may I have the usbx_host_controllers source code in the usbx\common folder, but there is no usbx_device_controllers file. Why?

bhnaphade, 2021-04-14:
Hello,
Which board are you targeting for the project?

geek-slt, 2021-04-14:
Hello，
My target MCU is STM32H743XIT6

bhnaphade, 2021-04-14:
Hi,
ST has added USBX support for their STM32H7 family MCUs: https://www.st.com/en/embedded-software/x-cube-azrtos-h7.html
Please refer the link for examples and driver code.

geek-slt, 2021-04-15:
I know that ST supports USBX and found relevant examples, but I don’t understand that USBX has a usbx_host_controllers file, but there is no usbx_device_controllers. Is it because the device_control is made by the chip manufacturer?

geek-slt, 2021-04-15:
And the files in usbx_stm32_host_controllers are not consistent with usbx_host_controllers

goldscott, 2021-04-15:
@geek-slt - Correct. Microsoft is not going to develop/maintain USBX drivers. Our partners like ST, NXP, Renesas, Microchip, etc. will develop and maintain USBX drivers. They know their devices much better than we do, so they can write drivers that take full advantage of all the hardware capabilities. See issue https://github.com/azure-rtos/usbx/issues/10 for a similar discussion.

closed 2021-04-15 by goldscott

geek-slt, 2021-04-16:
Understand    Thankyou verymuch