eclipse-threadx/usbx #14: duplicate line in : usbx/common/core/src/ux_host_stack_class_device_scan.c

parthindia47, 2021-05-21:
It might be big issue but there is duplicate line in : usbx/common/core/src/ux_host_stack_class_device_scan.c

```
    /* Perform the command initialization.  */
    class_command.ux_host_class_command_request      =   UX_HOST_CLASS_COMMAND_QUERY;
    class_command.ux_host_class_command_container    =   (VOID *) device;
    class_command.ux_host_class_command_vid          =   device -> ux_device_descriptor.idVendor;
    class_command.ux_host_class_command_pid          =   device -> ux_device_descriptor.idProduct;
    class_command.ux_host_class_command_class        =   device -> ux_device_descriptor.bDeviceClass;
    class_command.ux_host_class_command_subclass     =   device -> ux_device_descriptor.bDeviceSubClass;
    class_command.ux_host_class_command_subclass     =   device -> ux_device_descriptor.bDeviceSubClass;
    class_command.ux_host_class_command_protocol     =   device -> ux_device_descriptor.bDeviceProtocol;
    class_command.ux_host_class_command_iad_class    =   0;
    class_command.ux_host_class_command_iad_subclass =   0;
    class_command.ux_host_class_command_iad_protocol =   0;
```

bhnaphade, 2021-05-21:
Thank you for reporting the bug. We will fix this.

yuxin-azrtos, 2021-06-03:
The duplicated line is removed: https://github.com/azure-rtos/usbx/blob/62802608052e632026fd409f4c3af85be47c04db/common/core/src/ux_host_stack_class_device_scan.c#L100

closed 2021-06-03 by yuxin-azrtos