eclipse-threadx/usbx #15: Multi storage media creation

FreeOxygen, 2021-05-24:
In the **ux_host_class_storage_device_initialize.c** file, after defining the **UX_HOST_CLASS_STORAGE_NO_FILEX** macro, when creating storage_media, traversing ux_host_class_media will create multiple storage_media!, 
which should **break** at **line 209**.

```
#if defined(UX_HOST_CLASS_STORAGE_NO_FILEX)

        /* Find a free media slot for inserted media.  */
        storage_media =  (UX_HOST_CLASS_STORAGE_MEDIA *) class_inst -> ux_host_class_media;
        for (inst_index = 0; inst_index < UX_HOST_CLASS_STORAGE_MAX_MEDIA;
            storage_media ++, inst_index++)
        {

            /* Skip used storage media slots.  */
            if (storage_media -> ux_host_class_storage_media_storage != UX_NULL)
                continue;

            /* Use this free storage media slot.  */
            storage_media -> ux_host_class_storage_media_storage = storage;

            /* Save media information.  */
            storage_media -> ux_host_class_storage_media_lun = (UCHAR)storage -> ux_host_class_storage_lun;
            storage_media -> ux_host_class_storage_media_sector_size = (USHORT)storage -> ux_host_class_storage_sector_size;
            storage_media -> ux_host_class_storage_media_number_sectors = storage -> ux_host_class_storage_last_sector_number + 1;

            /* Invoke callback for media insertion.  */
            if (_ux_system_host -> ux_system_host_change_function != UX_NULL)
            {

                /* Call system change function.  */
                _ux_system_host ->  ux_system_host_change_function(UX_STORAGE_MEDIA_INSERTION,
                                    storage -> ux_host_class_storage_class, (VOID *) storage_media);
            }
        }
#endif
```

xiaocq2001, 2021-05-26:
Thank you for the feedback. We will update the file.

xiaocq2001, 2021-08-04:
Fixed in 6.1.8

closed 2021-08-04 by FreeOxygen