eclipse-threadx/usbx #16: Update usbx/common/usbx_host_classes/inc/ux_host_class_video.h as per latest UVC 1.5 specification.

parthindia47, 2021-05-31:
Hi,

It seems header file for video host class is not matching UVC 1.5 specification < https://www.usb.org/document-library/video-class-v15-document-set >.
for example ux_host_class_video.h mentions:

/* Define Video Class specific VS .  */

```
#define UX_HOST_CLASS_VIDEO_VS_UNDEFINED                                                    0x00
#define UX_HOST_CLASS_VIDEO_VS_INPUT_HEADER                                                 0x01
#define UX_HOST_CLASS_VIDEO_VS_OUTPUT_HEADER                                                0x02
#define UX_HOST_CLASS_VIDEO_VS_STILL_IMAGE_FRAME                                            0x03
#define UX_HOST_CLASS_VIDEO_VS_FORMAT_UNCOMPRESSED                                          0x04
#define UX_HOST_CLASS_VIDEO_VS_FRAME_UNCOMPRESSED                                           0x05
#define UX_HOST_CLASS_VIDEO_VS_FORMAT_MJPEG                                                 0x06
#define UX_HOST_CLASS_VIDEO_VS_FRAME_MJPEG                                                  0x07
#define UX_HOST_CLASS_VIDEO_VS_FORMAT_MPEG2TS                                               0x0A
#define UX_HOST_CLASS_VIDEO_VS_FORMAT_DV                                                    0x0C
#define UX_HOST_CLASS_VIDEO_VS_COLORFORMAT                                                  0x0D
#define UX_HOST_CLASS_VIDEO_VS_FORMAT_FRAME_BASED                                           0x10
#define UX_HOST_CLASS_VIDEO_VS_FRAME_FRAME_BASED                                            0x11
#define UX_HOST_CLASS_VIDEO_VS_FORMAT_STREAM_BASED                                          0x12
```

But there are other subtypes as well on page 172 , UVC 1.5 specification < https://www.usb.org/document-library/video-class-v15-document-set >.
Ideally following should be added :

```
#define UX_HOST_CLASS_VIDEO_VS_FORMAT_H264                                          0x13
#define UX_HOST_CLASS_VIDEO_VS__FRAME_H264                                          0x14
#define UX_HOST_CLASS_VIDEO_VS_FORMAT_H264_SIMULCAST                                0x15
```
This is just an example entire file need an update.

xiaocq2001, 2021-06-02:
Thanks for the feedback.
We will update the file for 1.5 spec.

xiaocq2001, 2021-08-04:
Updated in 6.1.8

parthindia47, 2021-08-09:
thanks I will test this

closed 2021-08-09 by parthindia47