eclipse-threadx/usbx #17: usbx on renesas rx mcu

casju1, 2021-06-07:
Hi,
I am updating usbx from v5.9 to v6.1.6 on a Rx63 mcu. Currently i am using the "USBX host rx Controller" but is it possible to change it to ohci/ehci host controller? Because i can see in your latest release that you does not include rx controller files anymore. I have tried changing it to ehci but it doesn't seem to work.

xiaocq2001, 2021-06-09:
Thanks for choosing USBX.
For the Renesas controller, we are still working on it and it will be available with Renesas release packet.

casju1, 2021-06-09:
Thanks for the answer.
Where do i find the Renesas release packet and when do you expect it will be implemented?

xiaocq2001, 2021-06-10:
I think Renesas will put his packets here: https://www2.renesas.cn/cn/en/design-support/software
They will put USBX and controller driver inside their packets, and they have their release plan.

Note if you have some previous packet that includes USBX 5.9 with host controller driver (HCD), you can try that HCD in USBX 6.

closed 2021-06-10 by casju1