eclipse-threadx/usbx #19: Recommendations for reducing USBx host stack size.

parthindia47, 2021-06-23:
Hi,
I downloaded sample code from : https://github.com/azure-rtos/samples for STM32.
In this one “sample_usbx_host_mass_storage” uses stack memory of 64KB:

```
#define USBX_MEMORY_SIZE            (64 * 1024)
ux_system_initialize(memory_pointer, USBX_MEMORY_SIZE, UX_NULL, 0);
```

I tried to test this with our hardware and I was able to get this working with 48KB.
But any reduction is stack memory below that generates “UX_MEMORY_INSUFFICIENT” error.

USBx documentation mentions this < https://docs.microsoft.com/en-us/azure/rtos/usbx/usbx-host-stack-2 >

Target Considerations
USBX requires between 24 KBytes and 64 KBytes of Read Only Memory (ROM) on the target in host mode. The amount of memory required is dependent on the type of controller used and the USB classes linked to USBX. Another 32 KBytes of the target's Random Access Memory (RAM) are required for USBX global data structures and memory pool. This memory pool can also be adjusted depending on the expected number of devices on the USB and the type of USB controller. The USBX device side requires roughly 10-12 K of ROM depending on the type of device controller. The RAM memory usage depends on the type of class emulated by the device.

I tried to test this with our hardware and I was able to get this working with minimum stack of 48KB.
But any reduction is stack memory below that generates “UX_MEMORY_INSUFFICIENT” error.

We only want to connect one USB device to a host at a time. And I understand that I can adjust parameters in "ux_port.h" to improve this code size. ( Though I had not much success ).

My Questions : 
**Q 1: Is there any list of steps/options/recommendation/link time define switch to try to reduce stack size of USBx ?

Q2: And is 32KB minimum stack required for USBx mode and is it possible to go below that ? if we only want to connect one device to a host let say mass storage device.**

Regards,
Parth

xiaocq2001, 2021-06-24:
For host stack with storage following could be adjusted:

For controller driver optimization (one controller only, TD/EDs are passed to controller driver and the actual size is related to controller driver implement):
UX_MAX_HCD=1
UX_MAX_ISO_TD=0
UX_MAX_TD=20
UX_MAX_ED=70

For core stack optimization (one device, one class only):
UX_MAX_CLASS_DRIVER=1
UX_MAX_DEVICES=1

For mass storage settings (one storage device and one LUN only, minimize buffer to just one sector):
UX_HOST_CLASS_STORAGE_MEMORY_BUFFER_SIZE=512
UX_HOST_CLASS_STORAGE_MAX_TRANSFER_SIZE=512
UX_HOST_CLASS_STORAGE_MAX_MEDIA=1

parthindia47, 2021-06-24:
Xiao , Thanks for suggestions. This really helped us to reduce stack memory size to 32KB.
Is it possible to go below 20KB ? if I have just one mass storage device attached to one host at a time.

Also , Can you please tell me what are full forms of ED , TD and TT. And can you please explain significance of them ?
and why we need separate Isochronous TD ? 

Is it possible to adjust following parameters ?
#define UX_MAX_DEVICE_ENDPOINTS                             6
#define UX_MAX_DEVICE_INTERFACES                            6
#define UX_MAX_SLAVE_INTERFACES                             16
#define UX_MAX_ROOTHUB_PORT                                 4
#define UX_MAX_TT                                                        8

#define UX_HCD_STM32_MAX_NB_CHANNELS                            12

xiaocq2001, 2021-06-25:
ED, TD is origin in EHCI/OHCI spec., means endpoint descriptor, transfer descriptor, in STM32 case, there is actually no TD but ED should be available for each allocated endpoints, in case we support single storage device, EP0 and two bulk endpoints are needed, so number of ED should be at least 3.
TT is for split transfer, which is only used when a low speed/full speed device connected through a high speed hub, not for your case.

The following is device stack parameters but not for host:
#define UX_MAX_DEVICE_ENDPOINTS 6
#define UX_MAX_DEVICE_INTERFACES 6
#define UX_MAX_SLAVE_INTERFACES 16

The following is for host and slightly effects RAM usage (only one root hub port, TT not used in case UX_MAX_DEVICES is 1) :
#define UX_MAX_ROOTHUB_PORT 4
#define UX_MAX_TT 8

The following is for host controller driver (HCD) each channel for an endpoint in current STM32 implement:
#define UX_HCD_STM32_MAX_NB_CHANNELS 12

parthindia47, 2021-06-27:
Hi Xiao,

When I reduced number of ED to 3 , function "_ux_hcd_stm32_periodic_tree_create" returns "UX_NO_ED_AVAILABLE".
I see there is hard coded value of 32 , does that mean I can not reduce my value of TD below 32 ? 

Also , can you please let me know , is it really possible to reduce USBx stack size to 16KB ?

xiaocq2001, 2021-06-30:
In STM32 latest examples (generated from STMCube) the HCD has been updated to use STM32 USB HAL and updated for the specific hardware, where periodic tree and many other things are optimized. You can get smaller stack size then.

parthindia47, 2021-07-06:
Hi xiao,
I used STM32 to generate USB code , but it is not using USBx stack ? 
am I missing something ?

Regards,
Parth

xiaocq2001, 2021-07-07:
In STM32Cube, in example selector, the name with Ux_.... are examples with USBX.

parthindia47, 2021-07-12:
hello xiao ,

thanks for your suggestion regarding mass storage class , do you have any suggestion to optimise stack for video class ?

xiaocq2001, 2021-07-16:
Just keep core stack optimized. Video class has no optimization flags yet.
Note for video class, there is video device configuration descriptors saved (2349 bytes for one of my video cam) for later use, there are transfer requests (UX_HOST_CLASS_VIDEO_TRANSFER_REQUEST_COUNT) allocated for transfers but it's necessary to set it to 8 or more for HIGH Bandwidth support. In application there must be frame buffers allocated for video streaming. That's the main video memory usage that can not be shrunk.

parthindia47, 2021-08-09:
Thanks Xiao for providing all the info , we can close the ticket for now

closed 2021-08-09 by parthindia47