eclipse-threadx/usbx #21: USB host stack sizes need to be large.

maxkunes, 2021-06-25:
Hi,

It seems the default stack size for UsbX in the gnu port file is too small for even my basic application of USB mouse and keyboard. I believe it is set to 2048 by default. 8192 seems to do the trick for me.

When the stack size is too small I always get a hard fault in the UX function that searches for a free memory block called by the ux allocate function. 

It's not really a bug, but it seems something has changed recently, from the older versions, that are requiring a larger stack size.

parthindia47, 2021-06-25:
Max,
Is this stack size for threads or is it stack size for USBx ?

Thread size is provided here
tx_thread_create(&tx_mass_storage_thread, "tx demo", mass_storage_thread_entry, 0,
          stack_pointer, DEMO_STACK_SIZE,
          30, 30, 1, TX_AUTO_START);

USBx stack size is passed in this function call :
`ux_system_initialize(memory_pointer, USBX_MEMORY_SIZE, UX_NULL, 0);`

And I would be surprised if this works with 8K stack.

maxkunes, 2021-06-25:
I'm referring to the thread stack sizes inside of usbx. They can be
configured in ux_user or ux_port.h

I'm not talking about the RAM give to ux in initialization.

wetechautomation, 2021-11-10:
I am able to get usbx working in 20KB in host mode stack

yuxin-azrtos, 2022-02-08:
@maxkunes The intention is to allow developers to specify a suitable stack size in ux_port.h.  Given different architectures we support, and the individual use case, its hard to set a default that is big enough but not too big to waste RAM space.