eclipse-threadx/usbx #22: Incorrect trace values

sbertin-telular, 2021-06-25:
These 2 both have a value of 855.
#define UX_TRACE_DEVICE_STACK_CONFIGURATION_GET                         (UX_TRACE_DEVICE_STACK_EVENTS_BASE + 5)             /* I1 = configuration value                                                                         */
#define UX_TRACE_DEVICE_STACK_CONFIGURATION_SET                         (UX_TRACE_DEVICE_STACK_EVENTS_BASE + 5)             /* I1 = configuration value                                                                         */

Within TraceX, 856 shows as Event: ux_trace_device_stack_configuration_set (856). The actual event was UX_TRACE_DEVICE_STACK_CONNECT, so I think UX_TRACE_DEVICE_STACK_CONFIGURATION_SET and below are all off by 1.

jdeere5220, 2021-06-25:
@sbertin-telular thank you for the feedback we will look into this and get back to you.

xiaocq2001, 2021-08-04:
IDs are updated in USBX 6.1.8

sbertin-telular, 2021-08-04:
Thank yoiu.