eclipse-threadx/usbx #23: Are CDRW drives supported for writing/reading files?

maxkunes, 2021-06-25:
I've noticed various places in the code that say CDRW drives aren't supported but I've also seen in the same code actions being taken if the current device is a CDRW. Confusing to say the least.

If they aren't supported, what actions specifically would need to be taken to support them?

xiaocq2001, 2021-06-26:
CDRW driver can be recognized, but no FileX volume is opened for read/write.
So yes, if you want to read/write CDRW driver, you need a file system library that supports CDRW formats.

maxkunes, 2021-06-26:
> CDRW driver can be recognized, but no FileX volume is opened for read/write.
> So yes, if you want to read/write CDRW driver, you need a file system library that supports CDRW formats.

Thanks for the quick response. So after the device enumerates, I can simply read and write sectors using the _ux_host_class_storage_media_write/read API? Or is there something else I have to do here.

xiaocq2001, 2021-06-26:
You can use _ux_host_class_storage_media_write/read to write/read sectors.