eclipse-threadx/usbx #24: is it possible to read device isochronous endpoints using host interrupt endpoints?

wetechautomation, 2021-06-27:
We are developing an example where device continuously stream data through an isochronous IN endpoint. Now my question was as interrupt and isochronous have lot of similarities :

1. Max 1024 packets in USB HS
2. both are for bandwidth guarantee.
3. Host can poll device every micro frame.

can I use my host interrupt endpoint to configure is to read from device's isochronous endpoints ?

device isochronous endpoints is following :

```
===>Endpoint Descriptor<===
bLength: 0x07
bDescriptorType: 0x05
bEndpointAddress: 0x81 -> Direction: IN - EndpointID: 1
bmAttributes: 0x05 -> Isochronous Transfer Type, Synchronization Type = Asynchronous, Usage Type = Data Endpoint
wMaxPacketSize: 0x0180 = 1 transactions per microframe, 0x180 max bytes
bInterval: 0x01
```

I understand there is NACK/ACK token in the interrupt endpoints , but let say if device ignores this NACK/ACK token can I still communicate with device using my interrupt endpoints ?

xiaocq2001, 2021-06-30:
No. Interrupt transfer has handshake phase but isochronous transfer has not. There is error if host expects handshake but device has no response.

wetechautomation, 2021-06-30:
No actually I mentioned reverse , let say my host have interrupt endpoint and I want to communicate with device isochronous Endpoint ? And now my host sends ACK after data , and device being isochronous endpoints ignores that. can I keep doing this ?

xiaocq2001, 2021-07-01:
Not recommended.

wetechautomation, 2021-07-05:
hello xiao,

in ux_host_class_hub_change_process.c file , can you please explain why length '0' is being send on line 109 and 110 ?

    /* The actual length should be cleared for the next transfer.  */
    transfer_request -> ux_transfer_request_actual_length =  0;

xiaocq2001, 2021-07-07:
I think it's something for backsword compatibility in case lower level driver does not reset actual length to 0 (actually that should be done by HCD but not in stack).

closed 2021-07-12 by wetechautomation