eclipse-threadx/usbx #25: USB HID class : Composite device does not pass compliance tests.

t-yabui, 2021-07-01:
Hi,
I made a composite device that include multiple HID devices.
It is recognized by Windows and works. However it does not pass HID Tests with the USB2CV test tool.

* HID Class Descriptor Test
* HID Class GET/SET Idle Test
* HID Class Report Descriptor Test

The tests uses GET_DESCRIPTOR with ``wIndex`` parameter to get HID descriptors.
However current code does not use the parameter, therefore it does not return a HID descriptor except the first one.

I think that It should support ``request_index`` parameter as follows:

[ux_device_class_hid_descriptor_send.c.patch.txt](https://github.com/azure-rtos/usbx/files/6745214/ux_device_class_hid_descriptor_send.c.patch.txt)

xiaocq2001, 2021-07-01:
Thanks for the feedback. We will fix in later updates.

xiaocq2001, 2021-08-04:
Updated in 6.1.8

t-yabui, 2021-08-30:
Thank you for handling the matter.

closed 2021-08-30 by t-yabui