eclipse-threadx/usbx #26: The ux_host_class_cdc_ecm not support process zero-length packet

zxslzx, 2021-07-27:
On linux we can use URB_ZERO_PACKET flag, but usbx doesn't seem to support similar operations.

xiaocq2001, 2021-08-02:
Is there any special reason to use URB_ZERO_PACKET ?
USBX binds NetX and all network operations are handled by NetX or triggered by NetX operations, so I don't thing the API can be exactly same as linux API.

zxslzx, 2021-08-04:
> Is there any special reason to use URB_ZERO_PACKET ?
> USBX binds NetX and all network operations are handled by NetX or triggered by NetX operations, so I don't thing the API can be exactly same as linux API.

On _Universal Serial Bus Communications Class Subclass Specification for Ethernet Control Model Devices_ 
[image]

xiaocq2001, 2021-08-04:
In USBX 6.1.4 the segmentation is already handled.

closed 2021-08-04 by zxslzx