eclipse-threadx/usbx #27: Differences between STM32F1 and STM32H7 series MCUs using USBX

arilink, 2021-07-29:
I have successfully run the ThreadX+USBX program on the STM32H7 chip. The USB composite device (CDC ACM+HID) is used, so I tried to implement the same function on STM32F105RBT6, but I ran into trouble, running USB on F105 The composite device can run normally at the beginning, but after 40s, the USB will be disconnected and reconnected, and the USB will not be recognized. If I use HID or CDC alone, everything works fine . I learned that H7 has 9 endpoints and F1 has only 4 USB endpoints. Will it? Is USB working abnormally because of the number of endpoints?

xiaocq2001, 2021-07-30:
It's possible number of endpoints issue.
I checked the example generated from STMCube for STM32H7, the following endpoints are used: 0x01, 0x82, 0x83 for CDC and 0x84 for HID, total 5 endpoints including EP0. If that's directly used in STM32F105 project, there is problem running endpoint 0x84 (as you mentioned only 4 USB endpoints are supported by F1).

arilink, 2021-07-30:
In this case, my guess is correct. Is there a way to reduce the number of endpoints occupied by CDC? For example, cancel the control endpoint 0x01 of CDC?

arilink, 2021-07-30:
It's really strange. CubeMX generates the USB driver of F103. hpcd_USB_OTG_FS.Init.dev_endpoints = 8 and F105, which is more advanced than F103, has only 4 endpoints. This is too outrageous.

closed 2021-07-31 by arilink

flamingvines2, 2021-09-01:
Hi Arilink,
Do you mind sharing your H7 codes? I am trying to port to H753 with STMCube but so far I can't make it work.
Thanks so much

arilink, 2021-09-01:
@flamingvines2 HID+CDC？

flamingvines2, 2021-09-01:
Yes, please. Thanks a lot!

arilink, 2021-09-01:
leave me a email , the file size is very big ,github can‘t support to upload

***@***.***
 
From: flamingvines2
Date: 2021-09-01 15:38
To: azure-rtos/usbx
CC: arilink; State change
Subject: Re: [azure-rtos/usbx] Differences between STM32F1 and STM32H7 series MCUs using USBX (#27)
Yes, please. Thanks a lot!

flamingvines2, 2021-09-01:
vines2001@yahoo.com

flamingvines2, 2021-09-01:
I think Github censored the email. I will a space in between. My email is

v i n e s 2 0 0 1 @ y a h o o . c o m

arilink, 2021-09-01:
Did you receive the file I sent?

***@***.***

flamingvines2, 2021-09-01:
Yes I did. Thanks a lot. I am looking at it and comparing it with what I did. It seems that my code is almost identical to what you are doing except for some HAL IO writes example - HAL_GPIO_WritePin(GPIOI, GPIO_PIN_8, GPIO_PIN_SET). Probably you are just monitoring something.

As I am not using Arm Keil, do you mind telling me what exact chipset you are using? STM32H7?? Are you using a development board or custom board?

arilink, 2021-09-02:
I only operate IO to use LED lights to identify the running status of the system

I am using a board made by STM32H743XBH6

flamingvines2, 2021-09-02:
I solved the issue. It was a hardware issue not really on the firmware. Anyway, thanks.

arilink, 2021-09-03:
ok  congratulate