eclipse-threadx/usbx #28: STM32 F105 cannot use USBX CDC ACM normally

arilink, 2021-07-31:
I have successfully ported ThreadX and USBX to STM32F105, and USBX HID can work normally. When I try to use USBX CDC ACM, the problem occurs. There is VCOM virtualized by STM32 in the device manager of windows, but when I try to use this serial port STM32 will freeze and stop running during communication. Why is there such a problem? I do the same operation on STM32 H7 and there is no problem

xiaocq2001, 2021-08-02:
Are you running the application with debugger? To see where the program goes.

arilink, 2021-08-02:
I tried to add breakpoints to observe in Debug mode, but unfortunately, as soon as VCOM sends data, the system will be stuck and MDK has not found the stuck position.、

arilink, 2021-08-02:
I tried to upload the code here, but it failed to upload. Can you leave your email? I will communicate with you via email

arilink, 2021-08-02:
The current problem has been solved. CDC can be used to communicate normally. But a new problem has arisen, that is, I can't use high frequency to send data to VCOM, otherwise the software will be stuck.Where will the CDC transceiver performance be restricted?

closed 2021-08-02 by arilink