eclipse-threadx/usbx #29: STM32F105 VCOM abnormality occurs after accumulatively receiving a certain length of data

arilink, 2021-08-02:
I transplanted ThreadX+USBX CDC Device on STM32F105, and then I used the serial debugging assistant to send data to VCOM. When the cumulative amount of data sent exceeds 500 bytes, it will malfunction and cannot continue to send, and the program can always be sent out through VCOM. Why?

xiaocq2001, 2021-08-03:
Not sure what's your application flow for data reading from host, but a possible case is, there is some issue in step 2 in following application sequence:
1. data is read
2. data is forwarded to your serial port
3. another read started
If there is some issue in step2 (as far as I know in ST demo when there is any error it halts inside thread), step3 is not issued and the device is not able to accept further data from host.

arilink, 2021-08-03:
@xiaocq2001 Indeed, as you said, the data is read and sent to the serial port via DMA, but I don’t understand why there is a problem here. The code cannot be uploaded. Can we communicate by email? I will send you the program to check

arilink, 2021-08-03:
I shielded the code related to serial port forwarding data. The problem still exists
On F105RBT6, USBX has problems with the data processing (OUT) sent by the host. The specific manifestation is HAL_PCDEx_SetRxFiFo(&hpcd_USB_OTG_FS, 0x200) when the USB is initialized; the RX FIFO is 512 bytes. When the host sends more than 512 data to the STM32 through VCOM After the byte, VCOM will stop working. The same situation also occurs on HID devices. When the data received by app_usbx_device_thread_hid_callback() exceeds 512 bytes, it will also stop working. The same code will not have problems on H7.
My guess is that USBX did not correctly read data from the RX FIFO on the STM32 F105, which caused the RX FIFO to overflow and freeze, but I don’t know how to fix it. Can you provide some information?

arilink, 2021-08-03:
@xiaocq2001 My email is shiliutao@arilink.com If you have time, please feel free to contact me. This problem has troubled me for a long time.

xiaocq2001, 2021-08-04:
From the description it's something related to FIFO configuration inside F105, so my suggestion:
1. 512 bytes seems too large for FS 64 bytes packets and can be smaller. The total FIFO size is limited for F105 please check your total FIFO usage in your project.
2. USBX runs on top of STM HAL and hardware, your issue seems related to hardware and STM32 HAL, so if you contact ST you may get more, since they know those better.

arilink, 2021-08-04:
1.The total USB FIFO of F105 is 1.25K. My current allocation of FIFO is as follows:
 /* Set Rx FIFO */
   HAL_PCDEx_SetRxFiFo(&hpcd_USB_OTG_FS, 0x200);
   /* Set Tx FIFO 0 */
   HAL_PCDEx_SetTxFiFo(&hpcd_USB_OTG_FS, 0, 0x10);
   /* Set Tx FIFO 2 */
   HAL_PCDEx_SetTxFiFo(&hpcd_USB_OTG_FS, 2, 0x10);
   /* Set Tx FIFO 3 */
   HAL_PCDEx_SetTxFiFo(&hpcd_USB_OTG_FS, 3, 0x80);
2. I haven't contacted the ST developer at the moment, can I send the code to you for checking?

xiaocq2001, 2021-08-04:
Something form ST datasheet:

RXFD: RxFIFO depth
This value is in terms of 32-bit words.
Minimum value is 16
Maximum value is 256

0x200 means 512 * 4 = 2K bytes which exceeds limit!

arilink, 2021-08-04:
I got it, I understood it wrong, thank you for your guidance

closed 2021-08-04 by arilink