eclipse-threadx/usbx #30: Multiple transaction in single microframe for interrupt endpoints

parthindia47, 2021-08-09:
Hi,

I am using STM32 mass storage HOST example from : https://github.com/azure-rtos/sample

I was going through code ( **in STM32 mass storage HOST example** ) and in : 
`UINT  _ux_hcd_stm32_td_schedule(UX_HCD_STM32 *hcd_stm32, UX_HCD_STM32_ED *ed)`

I seen this line :  
`number_packets =  (td -> ux_stm32_td_length + endpoint -> ux_endpoint_descriptor.wMaxPacketSize - 1) / endpoint -> ux_endpoint_descriptor.wMaxPacketSize;`

Now if you see carefully then this line will always return 1 irrespective of ux_endpoint_descriptor.wMaxPacketSize.

In my opinion it should be : 
```
max_packet_size  = endpoint -> ux_endpoint_descriptor.wMaxPacketSize & UX_MAX_PACKET_SIZE_MASK;
number_packets =  (td -> ux_stm32_td_length + max_packet_size - 1) / max_packet_size;
```

Not sure fully , but let me know your thoughts on this.
Also I wanted to know does USBx code supports multiple transaction in single microframe for interrupt endpoints ?

and also multiple transaction in single microframe for isochronous endpoints ? I simply tried to search "UX_HCD_STM32_OTG_FS_HCTSIZ_DPID_MDATA" but was not able to find it ?

xiaocq2001, 2021-08-11:
Thanks for the feedback.
Unfortunately the HCD inside the sample is obsoleted.
Please try the HCD which is released by ST in STMCube, with STM32 H7 examples (can be generated by STM32Cube for UX examples), where STM32 HAL is used to support STM32 controllers.

parthindia47, 2021-08-11:
Hi xiao,

Thanks for the suggestion , STM32 cube code is very optimized compared to sample code.
Though I have same question. Is multiple transactions supported in single microframe for isochronous transfer ?
I see following code,
when do we load HC_PID_DATA1 and HC_PID_DATA2 ? or is it maintained by USB IP ?

```
in stm32h7xx_hal_hcd.c
    
case EP_TYPE_ISOC:
      hhcd->hc[ch_num].data_pid = HC_PID_DATA0;
      break;
```

xiaocq2001, 2021-08-12:
For ISO transactions, the PID sequence is fixed for high-bandwidth case, it's possible the PIDs are generated automatically by hardware and the setting actually has no effect in this case. 
But you should ask ST for the actual behavior of their USB controller IP, also the stm32h7xx_hal_hcd.c code (that's offered by ST).