eclipse-threadx/usbx #31: Is it possible to cancel the CDC notification endpoint？

arilink, 2021-08-24:
Hello, I try to use ThreadX+USBX in my project. My USB needs CDC and HID, but my MCU only has 4 endpoints (including EP0), so there are only 3 endpoints available for me to use. CDC occupies 3 endpoints, which makes the endpoints not enough to be used. I tried to block the notification endpoint of CDC and assign it to HID. It is possible to run ThreadX+USBX (CDC+HID) alone, but when I add other components The system will crash. I tried ThreadX+USBX (HID only) to add other components and it works normally. The problem lies in the CDC. How can I solve it?

xiaocq2001, 2021-08-24:
Guessing you are implementing device.

For CDC ACM serial bridge, at least for windows, the CDC notification endpoint can not be removed (windows not accept such kind of device).

The possible solution is, to modify DCD, to bypass physical management of CDC interrupt IN notification endpoint, so it's enough to manage 4 physical endpoints for control endpoint 0, HID interrupt IN, CDC bulk IN and OUT. You can bypass the endpoint by checking its endpoint address (controlled in application by endpoint descriptor). The APIs to bypass include endpoint creation, destroy, transfer request, there are some other endpoint related APIs but they are actually not used on CDC interrupt IN notification endpoint.

arilink, 2021-08-24:
Yes, I currently retain the control of CDC in the CDC descriptor, but instead of assigning an endpoint to it, I assigned the endpoint to HID IN so that both CDC and HID can work normally. But when I add other programs to the project, the system There will be an exception, I don’t know where the conflict occurred

xiaocq2001, 2021-08-24:
Are you assign same endpoint to different interfaces, I don't think that's allowed, since different interface needs different pipe for interrupt transfer. So you should report different endpoint addresses for CDC control interface and HID interface to make them work together.

The solution I mentioned tends to report good descriptor structure to host, but bypass the CDC interrupt endpoint to reserve the physical resource to HID.

arilink, 2021-08-25:
I think my current measures are consistent with what you described. Let me upload my current descriptor. Do you want to meet the requirements? I feel that this kind of problem should be encountered by many people
[USBlyzer Report.pdf](https://github.com/azure-rtos/usbx/files/7042878/USBlyzer.Report.pdf)

xiaocq2001, 2021-08-25:
The issue in your current descriptors is that, your CDC control interface has no interrupt endpoint description. This generates issue while working with standard windows drivers. The interface must have one endpoint as interrupt endpoint. That follows section 3.6.2.1 CDC spec 1.1, one of notifications is required for CDC-ACM serial emulation.

My way tends to report the descriptors as standard, but skip the CDC notification endpoint operations at physical level to avoid issues on host driver side.

arilink, 2021-09-01:
@xiaocq2001 This problem is sloved, Thank you Very much

closed 2021-09-01 by arilink