eclipse-threadx/usbx #32: Can Filex and USBX work on one storage medium at the same time？

arilink, 2021-09-01:
I have a question. I want to use an SD card or Flash as a storage medium, and then make a USB MSG device so that the PC can read and write the files in it. At the same time, STM32 uses FileX to access the files inside. Is this allowed?

xiaocq2001, 2021-09-01:
No, it's not recommended. USB MSC device modifies sectors which makes conflicts with FileX.

zhangshengping, 2021-09-10:
> No, it's not recommended. USB MSC device modifies sectors which makes conflicts with FileX.

is any plan to imply mtp class device?
@xiaocq2001

xiaocq2001, 2021-09-10:
Please try PIMA class, with UX_PIMA_WITH_MTP_SUPPORT defined.

arilink, 2021-09-15:
@xiaocq2001 If it is not operated at the same time, the system starts FileX to operate the SD card at the beginning of the system, and starts USB MSC to update the files on the SD card when needed. FileX only reads the files. Can this avoid conflicts?

xiaocq2001, 2021-09-15:
No, FileX may cache something that could be updated by USBX.
Recommended sequence:
1. Close media (FileX)
2. Start USBX
3. host may operate the media
4. Stop USBX
5. Open media (FileX)

arilink, 2021-09-15:
Ok，i will try as you said

arilink, 2021-09-15:
@xiaocq2001 I take the liberty to ask, are you Chinese?

xiaocq2001, 2021-09-15:
Yes.

arilink, 2021-09-16:
Can you leave WeChat? Want to learn more from you

derskop, 2021-09-22:
hey,sir,can you give me a project about the usbx host msc, recently  i  am studying  how to move it to my project   meeting some diffcult. thanks.

xiaocq2001, 2021-09-22:
Maybe you can check https://github.com/azure-rtos/samples.
Or you can try to generate USBX host storage demo from STM32 Cube.