eclipse-threadx/usbx #34: Does USBX MSC support SPI FLASH?

arilink, 2021-09-15:
I have tried to use the SD card as a storage device on the USBX MSC, and then I wrote a program to read the external Nor FLASH through SPI, but when I used the USB MSC with the Nor Flash driver, the problem appeared. On windows Storage devices did not appear as expected. What are the possible reasons for the above phenomenon?

xiaocq2001, 2021-09-15:
Not sure "did not appear", do you mean no pop up on new volume?
MSC works with callbacks to access media, no matter it's SD card or SPI flash. If PC see the USB connection but failed to mount disk, maybe there is issue in application callback implement of reading/writing SPI flash.

arilink, 2021-09-15:
Yes, there is no pop-up window on the new volume. Let me check the callback problem of SPI flash. No relevant examples can be found on the Internet.

yuxin-azrtos, 2022-02-08:
Closing for no activity.

closed 2022-02-08 by yuxin-azrtos