eclipse-threadx/usbx #35: Not getting deactivate callback when USB is removed in usbx_device_mass_storage on IMXRT1060-EVK

shwetachauhan, 2021-09-16:
Hi,

I have modified usbx_device_mass_storage sdk example for IMXRT060-EVK to include the activate and deactivate callback as shown below:

    storage_parameter.ux_slave_class_storage_instance_activate =
        demo_thread_media_activate;
    storage_parameter.ux_slave_class_storage_instance_deactivate =
        demo_thread_media_deactivate;

As of now I am just printing inside these callbacks as shown below:
VOID demo_thread_media_activate(VOID *storage)
{
    PRINTF("demo_thread_media_activate.....\r\n");
}
VOID demo_thread_media_deactivate(VOID *storage)
{
    PRINTF("demo_thread_media_deactivate.....\r\n");
}

I do receive the activate callback when I plug in USB but do not get deactivate callback when plugging out the USB but gets both deactivate and activate callback prints after I plugin USB post plugging out .

xiaocq2001, 2021-09-17:
Not sure which SDK example you are using, but the DCD implement has some issue on disconnect handling here.

In ux_dcd_mcimx6_interrupt_thread.c, line 225 there is check if device state is attached, actually the check should be if device state is "above" attached including addressed or configured, so it should be something like following:

if (device -> ux_slave_device_state >= UX_DEVICE_ATTACHED) _ux_device_stack_disconnect();

Thus when disconnect detected after enumeration, stack is notified on disconnection and instance is deactivated.

yuxin-azrtos, 2021-10-11:
@shwetachauhan Does this answer your question?

closed 2022-02-08 by yuxin-azrtos