eclipse-threadx/usbx #39: HID IN data transmission frequency is only about 10 times per second

arilink, 2021-09-27:
I used ThreadX and USBX STM32 in the project as USB HID devices to connect to the computer. My program collects sensor data every millisecond and calls the ux_device_class_hid_event_set() function to send to the host computer, but the host computer can only receive about 10 per second The data, why is there such a situation, ux_device_class_hid_event_set is also set to 1ms in the endpoint descriptor

arilink, 2021-09-28:
1.I #define TX_TIMER_TICKS_PER_SECOND 1000 in the tx_user.h file
2.The delay of the descriptor is set to 1ms
[image]

arilink, 2021-09-28:
I used an oscilloscope to confirm that the time reference of tx_thread_sleep() is normal. Another project of mine uses TinyUSB. The descriptors of the two projects are the same. But when using USBX, the amount of token packet data observed through BUS hound is very low. I don’t understand why.

xiaocq2001, 2021-09-28:
1. Please confirm the ThreadX tick is 1ms (check ThreadX clocks settings), if that's not really 1ms changing the macro will cause problem.
2. The descriptor seems OK if you are using full speed.

arilink, 2021-09-28:
I measured it with an oscilloscope to determine that the ThreadX tick is 1ms, and the same descriptor can be used normally on STM32H7. It can transmit data 1000 times per second, but it does not work on STM32F1.

arilink, 2021-09-28:
I can basically determine that the USB transmission problem is caused by FIFO allocation. When I only use HID and allocate the USB FIFO as shown below,
  HAL_PCDEx_SetRxFiFo(&hpcd_USB_OTG_FS, 0x80);
  HAL_PCDEx_SetTxFiFo(&hpcd_USB_OTG_FS, 0, 0x40);
  HAL_PCDEx_SetTxFiFo(&hpcd_USB_OTG_FS, 1, 0x80);
 the transmission rate can reach 1000 times per second.
But my project needs to use HID and CDC at the same time. When I set the FIFO according to the following allocation, the transmission speed will be reduced. I want to know how to deal with the FIFO allocation to ensure that HID and CDC can work normally
  HAL_PCDEx_SetRxFiFo(&hpcd_USB_OTG_FS, 0x80);

  /* Set Tx FIFO 0 */
  HAL_PCDEx_SetTxFiFo(&hpcd_USB_OTG_FS, 0, 0x10);

  /* Set Tx FIFO 2 */
  HAL_PCDEx_SetTxFiFo(&hpcd_USB_OTG_FS, 2, 0x80);

  /* Set Tx FIFO 3 */
  HAL_PCDEx_SetTxFiFo(&hpcd_USB_OTG_FS, 3, 0x30);

arilink, 2021-09-29:
@xiaocq2001 Is there any solution? This is really confusing

xiaocq2001, 2021-09-29:
Not so familiar with ST FIFO setting.
But here are some suggestion I think worth to try:
1. Since you are not focused on reading from host, suggest to decrease RxFIFO, e.g., to 0x20.
2. Since you are not focused on CDC communication speed, you can decrease bulk FIFO, guessing FIFO 2, e.g., to 0x40.
3. Now you have enough space for interrupt FIFO, which can set to 0x80.

arilink, 2021-09-29:
Is the interrupt FIFO FIFO 0? When I only use HID, it can work normally even if the HID FIFO is set very small.
   HAL_PCDEx_SetRxFiFo(&hpcd_USB_OTG_FS, 0x40);
   HAL_PCDEx_SetTxFiFo(&hpcd_USB_OTG_FS, 0, 0x10);
   HAL_PCDEx_SetTxFiFo(&hpcd_USB_OTG_FS, 1, 0x10);
Once I join CDC and use USB as a composite device, the transmission frequency of HID will drop. I tried to adjust the FIFO size many times, but I couldn’t solve

xiaocq2001, 2021-09-29:
OK. From your descriptor, your endpoint is 0x81, then RxFIFO 1 is used for interrupt endpoint ...

arilink, 2021-09-29:
My HID IN endpoint is 0x83 CDC IN is 0x82 If TX FIFO1 is an interrupt endpoint, what is the purpose of configuring TX FIFO0?

xiaocq2001, 2021-09-29:
TX FIFO 0 is for control endpoint 0 IN.

arilink, 2021-09-29:
So no matter what device type is used, RX FIFO and TX FIFO 0 are necessary. My current CDC IN uses 0x82 endpoint and HID IN uses 0x83 endpoint. Therefore, I have configured TX FIFO 2 and TXFIFO 3 for this operation should be correct?

xiaocq2001, 2021-09-29:
Yes.

arilink, 2021-09-29:
This is very strange. Could it be that USBX does not support the STM32F1 series?

xiaocq2001, 2021-09-30:
That's related to the hardware peripheral usage.
Since the total FIFO is limited, user should control the FIFO for his purpose.

arilink, 2021-09-30:
The total FIFO of STM32F105RBT6 is 1.25K. My allocation is based on the needs. I allocate the same size of FIFO to STM32H750 and it can work normally.

xiaocq2001, 2021-09-30:
If same hardware configuration generates different results, please check with ST to see if there is hardware difference between two chips.

arilink, 2021-10-07:
I tried to report my current problems to the ST community, but I didn’t get any useful information. Can you contact the ST official?

arilink, 2021-10-14:
@liydu I am looking forward to your reply