eclipse-threadx/usbx #41: OHCI controller for full speed usb host

s-bruce13, 2021-11-17:
Hello,
I'm trying to get the example available running at full speed instead of high-speed. The example is the usbx_host_hid_keyboard for the LPC5528.
The keyboard enumerates correctly, but it doesn't respond to any key inputs. The only changes I saw were necessary in board_setup.c/.h which sets up the appropriate USB0, clocks, and interrupts. 

I added the ohci drivers in the project. However when calling _ux_host_class_hid_keyboard_key_get, I do not get any information.

Are there any other changes required for full-speed to work?

xiaocq2001, 2021-11-22:
Thanks for the feedback, we will check that.

xiuwencai, 2021-12-15:
There is an issue in the OHCI driver. The fix will be in the next USBX release and also in MCUXpresso SDK.

closed 2023-02-06 by xiaocq2001