eclipse-threadx/usbx #43: Does USBX support host class for CDC-ECM device or RNDIS device?

MichaelZhg, 2022-01-21:
Many IOT device using wireless network device to visit network,so USBX is or not support wireless device?

MichaelZhg, 2022-01-21:
I want to use USBX to realize the network access of wireless network device on STM32 platform,using CDC-ECM or RNDIS device to access.

goldscott, 2022-01-21:
USBX has a host CDC-ECM class here: https://github.com/azure-rtos/usbx/tree/master/common/usbx_host_classes/src

MichaelZhg, 2022-01-22:
> USBX has a host CDC-ECM class here: https://github.com/azure-rtos/usbx/tree/master/common/usbx_host_classes/src

Yeah,I have see it ,about RNDIS device support have any plain?

MichaelZhg, 2022-01-22:
> USBX has a host CDC-ECM class here: https://github.com/azure-rtos/usbx/tree/master/common/usbx_host_classes/src

Hi Glodscoot,
    Because we have a project that needs to use the STM32 USB host mode to drive the wireless network card, using the protocol mode of rndis or cdc-ecm, but there are too few reference examples of USBX, so we can only ask you for help.
    Look forward to your response,thank you!

AndyW999, 2022-04-04:
I would like this too so I can connect to other RF devices that require an RNDIS Host to connect to.
It is a glaring omission by Microsoft...

xiaocq2001, 2022-04-12:
There is no plan to support RNDIS on host side.

AndyW999, 2022-04-12:
Disappointing but not really a surprise...

closed 2023-01-30 by xiaocq2001