eclipse-threadx/usbx #44: How about a callback when new event is received in receiver thread

gawied, 2022-02-07:
First of all thank you vey much for adding support for optional OUT endpoint.

The current code structure calls for another thread to poll the ux_device_class_hid_receiver_event_get() function in order to be notified when new events are received data on the OUT endpoint.
How about adding a callback function, which can be part of the UX_DEVICE_CLASS_HID_RECEIVER structure, that will be called when the _ux_device_class_hid_receiver_thread determines that a new event was received. This will eliminate another polling thread and the system will be more event driven as well.

I certainly might have interpreted the new code incorrectly and would appreciate it if there are other ways than a polling thread to get notified of new received events.

ayedmo, 2022-02-07:
@gawied same idee, i think that is helpful to add a callback when received data from Host.

xiaocq2001, 2022-02-08:
Thanks for the suggestion, we will put that feature in our plan.

gawied, 2022-02-08:
Thanks that will be great!

Any idea on the time span of your plan? What are we talking about here - weeks, months?

yuxin-azrtos, 2022-02-08:
This callback will be added to the next release of Azure RTOS (towards the end of April 2022).

closed 2022-02-08 by yuxin-azrtos