eclipse-threadx/usbx #46: Example of device MSD+CDC

maxkunes, 2022-02-10:
I have an embedded device that I currently run a USB slave MSD on to access the internal filesystem. I would also like to have a serial communication so that I can use teraterm to talk and see output from this device on that same port. I know USBX supports both of these functionalities (I've tried both and they work), but I'm not skilled enough to develop a composite descriptor for this use case.

Is there any way someone could show what a composite descriptor would look like for a USB device that has MSD and CDC?

xiaocq2001, 2022-02-11:
Here is example for your reference: there is one configuration for the high speed device; there are 3 interfaces in the configuration; interface 0 and 1 for CDC, uses endpoint 0x81, 0x02, 0x83; interface 3 for storage, uses endpoint 0x04 and 0x85.

```c
/* Device Descriptor */
0x12,       /* bLength : 18 */
0x01,       /* bDescriptorType */
0x00, 0x02, /* bcdUSB : 0x0200 : USB 2.0 */
0xef,       /* bDeviceClass : 0xEF : MISC */
0x02,       /* bDeviceSubClass : 0x02 */
0x01,       /* bDeviceProtocol : 0x01 */
0x40,       /* bMaxPacketSize0 : 64 : 64 */
0x84, 0x84, /* idVendor : 0x8484 : Express Logic */
0x00, 0x00, /* idProduct */
0x00, 0x01, /* bcdDevice */
0x01,       /* iManufacturer */
0x02,       /* iProduct */
0x03,       /* iSerialNumber */
0x01,       /* bNumConfigurations */
  /* Device_Qualifier Descriptor */
  0x0a,       /* bLength */
  0x06,       /* bDescriptorType */
  0x00, 0x02, /* bcdUSB : 0x0200 : USB 2.0 */
  0x02,       /* bDeviceClass : 0x02 */
  0x00,       /* bDeviceSubClass : 0x00 : Reset */
  0x00,       /* bDeviceProtocol : 0x00 : Reset */
  0x40,       /* bMaxPacketSize0 : 64 : 64 */
  0x01,       /* bNumConfigurations */
  0x00,       /* bReserved */
  /* Configuration Descriptor, total 98 */
  0x09,       /* bLength */
  0x02,       /* bDescriptorType */
  0x62, 0x00, /* wTotalLength : 98 */
  0x03,       /* bNumInterfaces */
  0x01,       /* bConfigurationValue */
  0x00,       /* iConfiguration */
  0x40,       /* bmAttributes */
              /* D6 : 0x1 : Self-powered */
              /* D5, Remote Wakeup : 0x0 : Not supported */
  0x00,       /* bMaxPower : 0 : 0mA */
    /* Interface Association Descriptor */
    0x08,       /* bLength */
    0x0b,       /* bDescriptorType */
    0x00,       /* bFirstInterface */
    0x02,       /* bInterfaceCount */
    0x02,       /* bFunctionClass : 0x02 */
    0x02,       /* bFunctionSubClass : 0x02 */
    0x00,       /* bFunctionProtocol : 0x00 : Reset */
    0x00,       /* iFunction */
      /* Interface Descriptor */
      0x09,       /* bLength */
      0x04,       /* bDescriptorType */
      0x00,       /* bInterfaceNumber */
      0x00,       /* bAlternateSetting */
      0x01,       /* bNumEndpoints */
      0x02,       /* bInterfaceClass : 0x02 : CDC Control */
      0x02,       /* bInterfaceSubClass : 0x02 */
      0x01,       /* bInterfaceProtocol : 0x01 */
      0x00,       /* iInterface */
        /* CDC Header Functional Descriptor */
        0x05,       /* bLength */
        0x24,       /* bDescriptorType */
        0x00,       /* bDescriptorSubtype */
        0x10, 0x01, /* bcdCDC : 0x0110 : CDC 1.1 */
          /* CDC Abstract Control Management Functional Descriptor */
          0x04,       /* bLength */
          0x24,       /* bDescriptorType */
          0x02,       /* bDescriptorSubtype */
          0x0e,       /* bmCapabilities */
                      /* D3, notification Network_Connection : 0x1 : Supported */
                      /* D2, request Send_Break : 0x1 : Supported */
[... 93 lines cut ...]
      0x07,       /* bLength */
      0x05,       /* bDescriptorType */
      0x85,       /* bEndpointAddress */
                  /* D7, Direction : 0x01 */
                  /* D3..0, Endpoint number : 5 */
      0x02,       /* bmAttributes */
                  /* D1..0, Trasfer Type : 0x2 : Bulk */
                  /* D3..2, Synchronization Type : 0x0 : No Synchronization */
                  /* D5..4, Usage Type : 0x0 : Data endpoint */
      0x00, 0x02, /* wMaxPacketSize : 512 */
                  /* D10..0, Max Packet Size : 512 */
                  /* D12..11, Additional transactions : 0x00 */
      0x00,       /* bInterval : 0 : 0 */
```

When registering classes, you can refer to following code for CDC and storage:
```c
/* Register class for configuration value 1, interface 0 and 1 (CDC-ACM).  */
    status =  ux_device_stack_class_register(_ux_system_slave_class_cdc_acm_name, ux_device_class_cdc_acm_entry,
                                             1, 0, &cdc_acm_parameter);
/* Register class for configuration value 1, interface 2 (storage).  */
    status =  ux_device_stack_class_register(_ux_system_slave_class_storage_name, ux_device_class_storage_entry, 
                                                1, 2, (VOID *)&storage_parameter);

```

maxkunes, 2022-02-11:
Hmm, that setup doesn't seem to work for me. Windows doesn't seem to like it and the device keeps mounting and unmounting a serial port, but I never see the MSC device.

I've recorded a pcap file of the issues, in this dump, the relevant filter to see the USB transactions are "usb.src == "2.46.0".

Here is the file: https://easyupload.io/wbo0so (can't attach directly to this issue because of the file type)

My initialization code looks like this:

```
UINT status =
			_ux_device_stack_initialize(usb_cdc_msc_framework, sizeof(usb_cdc_msc_framework), usb_cdc_msc_framework, sizeof(usb_cdc_msc_framework),
										usbfs_string_framework, sizeof(usbfs_string_framework), usbfs_language_id_framework, sizeof(usbfs_language_id_framework), nullptr);

		if (status != UX_SUCCESS)
			return false;

		parameter.ux_slave_class_cdc_acm_instance_activate = cb_ux_slave_class_cdc_acm_instance_activate;
		parameter.ux_slave_class_cdc_acm_instance_deactivate = cb_ux_slave_class_cdc_acm_instance_deactivate;
		parameter.ux_slave_class_cdc_acm_parameter_change = cb_ux_slave_class_cdc_acm_parameter_change;

		/* Initilize the device storage class. The class is connected with interface
		 * 0 on configuration 1. */
		status = _ux_device_stack_class_register(_ux_system_slave_class_cdc_acm_name, ux_device_class_cdc_acm_entry, 1, 0, (VOID *)&parameter);

		if (status != UX_SUCCESS)
			return false;

		usbfs_storage_parameter.ux_slave_class_storage_parameter_number_lun = 1;

		/* Initialize the storage class parameters for reading/writing to the Flash
		 * Disk. */
		usbfs_storage_parameter.ux_slave_class_storage_parameter_lun[0].ux_slave_class_storage_media_last_lba = USBFS_LAST_BLOCK_INDEX;
		usbfs_storage_parameter.ux_slave_class_storage_parameter_lun[0].ux_slave_class_storage_media_block_length = USBFS_BLOCK_SIZE; // needs to be 512, not sure if this needs to match
		// filessytem
		usbfs_storage_parameter.ux_slave_class_storage_parameter_lun[0].ux_slave_class_storage_media_type = 0;
		usbfs_storage_parameter.ux_slave_class_storage_parameter_lun[0].ux_slave_class_storage_media_removable_flag = 0x80;
		usbfs_storage_parameter.ux_slave_class_storage_parameter_lun[0].ux_slave_class_storage_media_read = usbfs_media_read;
		usbfs_storage_parameter.ux_slave_class_storage_parameter_lun[0].ux_slave_class_storage_media_write = usbfs_media_write;
		usbfs_storage_parameter.ux_slave_class_storage_parameter_lun[0].ux_slave_class_storage_media_status = usbfs_media_status;
		usbfs_storage_parameter.ux_slave_class_storage_parameter_lun[0].ux_slave_class_storage_media_flush = usbfs_media_flush;

		/* Initilize the device storage class. The class is connected with interface
		 * 0 on configuration 1. */
		status = _ux_device_stack_class_register(_ux_system_slave_class_storage_name, _ux_device_class_storage_entry, 1, 2, (VOID *)&usbfs_storage_parameter);

		if (status != UX_SUCCESS)
			return false;

		/* Register the K64 USB device controllers available in this system */
		_ux_dcd_mcimx6_initialize(usb_device_base());

		/* Init the USB interrupt. */
		usb_device_interrupt_setup(USB_DEVICE_INTERRUPT_PRIORITY);
```

and the frameworks:

```

UCHAR usb_cdc_msc_framework[] = {
		/* Device Descriptor */
		0x12,       /* bLength : 18 */
		0x01,       /* bDescriptorType */
		0x00, 0x02, /* bcdUSB : 0x0200 : USB 2.0 */
		0xef,       /* bDeviceClass : 0xEF : MISC */
		0x02,       /* bDeviceSubClass : 0x02 */
		0x01,       /* bDeviceProtocol : 0x01 */
		0x40,       /* bMaxPacketSize0 : 64 : 64 */
		0x84, 0x84, /* idVendor : 0x8484 : Express Logic */
		0x00, 0x00, /* idProduct */
[... 155 lines cut ...]
		      0x02,       /* bmAttributes */
		                  /* D1..0, Trasfer Type : 0x2 : Bulk */
		                  /* D3..2, Synchronization Type : 0x0 : No Synchronization */
		                  /* D5..4, Usage Type : 0x0 : Data endpoint */
		      0x00, 0x02, /* wMaxPacketSize : 512 */
		                  /* D10..0, Max Packet Size : 512 */
		                  /* D12..11, Additional transactions : 0x00 */
		      0x00       /* bInterval : 0 : 0 */
};

UCHAR usbfs_string_framework[] = {
	/* Manufacturer string descriptor : Index 1 */
	0x09, 0x04, 0x01, 0x0c, 0x45, 0x78, 0x70, 0x72, 0x65, 0x73, 0x20, 0x4c, 0x6f, 0x67, 0x69, 0x63,

	/* Product string descriptor : Index 2 */
	0x09, 0x04, 0x02, 0x0a, 0x46, 0x6c, 0x61, 0x73, 0x68, 0x20, 0x44, 0x69, 0x73, 0x6b,

	/* Serial Number string descriptor : Index 3 */
	0x09, 0x04, 0x03, 0x04, 0x30, 0x30, 0x30, 0x31};

UCHAR usbfs_language_id_framework[] = {
	/* English. */
	0x09, 0x04};

```

maxkunes, 2022-02-11:
Hmm, I found one issue that seems to allow everything to work.

My version of USBX had the following defines in the support files (ux_cdc_mcimx6.h) :

```
#define UX_DCD_MCIMX6_MAX_ED                                           8	
#define UX_DCD_MCIMX6_MAX_ED_VALUE                               4
```

While it seems on a newer version of the library, they are set to this: (which seems correct, 8 bidir endpoints).

```
#define UX_DCD_MCIMX6_MAX_ED                                           16	
#define UX_DCD_MCIMX6_MAX_ED_VALUE                               8
```
I didn't catch this as both files say the same version;
`/*  05-19-2020     Chaoqiong Xiao           Initial Version 6.0           */`

But one of them is updated...

xiaocq2001, 2022-02-18:
Thanks for the information. 

From the macros, I guess the change tried to reduce managed number of endpoints to reduce memory usage, but for composite device the number of endpoints are increased (MSC needs 2 endpoints and CDC ACM needs 3 endpoints, plus 1 control endpoint).

yuxin-azrtos, 2022-02-22:
Closing.. Feel free to reopen it if you have additional questions.

closed 2022-02-22 by yuxin-azrtos