eclipse-threadx/usbx #47: About cdc-acm support in standalone mode

AlexYzhov, 2022-02-12:
Hi there, I recently noticed that usbx added standalone mode support in the latest v6.1.10_rel release, so I wrote some codes to initialize standalone usbx device stack as cdc-acm class.

But when I tried to build the firmware, the compiler warned undefined references to __ux_device_class_cdc_acm_read_ and __ux_device_class_cdc_acm_write_ functions.

Then I saw the implementation of these two function were all isolated by UX_DEVICE_STANDALONE and UX_DEVICE_STANDALONE macros.

My question is, are these functions deprecated in standalone mode, or just are lacked with support temporarily ? 
If these two functions can't be used for now, is there something instead?

And where can I go for further details like these problems, I will be glad to have a migration guide.

Great thanks for your help

xiaocq2001, 2022-02-14:
With RTOS, _read and _write can be done in thread blocking way, without RTOS (standalone) the functions that need polling should be used (_read_run and _write_run functions for device side),
While using _run functions, return code indicates how state machine is going (you can also check comments for the function implementation):

```c
/*    State machine Status to check                                       */
/*    UX_STATE_NEXT                         Transfer done, to next state  */
/*    UX_STATE_EXIT                         Abnormal, to reset state      */
/*    (others)                              Keep running, waiting         */
```

AlexYzhov, 2022-02-14:
I hacked line 117 in [ux_dcd_stm32_function.c](https://github.com/STMicroelectronics/x-cube-azrtos-h7/blob/main/Middlewares/ST/usbx/common/usbx_stm32_device_controllers/ux_dcd_stm32_function.c) :

`
status = (_ux_dcd_stm32_transfer_request(dcd_stm32, (UX_SLAVE_TRANSFER *) parameter) == UX_SUCCESS) ? (UX_STATE_NEXT) : (UX_STATE_ERROR);
`

And wrote a demo read function:

    /* reset transmission status */
    cdc_acm->ux_slave_class_cdc_acm_transmission_status = UX_FALSE;

    while (1)
    {
        uint32_t rlen = 0;
         UINT status = ux_device_class_cdc_acm_read_run(cdc_acm, &buf[0], len, &rlen);
         switch (status)
        {
            case UX_STATE_NEXT:
                return rlen;
            case UX_STATE_EXIT:
                return 0;
            default:
                break;
        }
    };

And it seems now data from OUT endpoint can be read into usbx internal buffer _transfer_request->ux_slave_transfer_request_data_pointer_ by reading back memories:
[image]

xiaocq2001, 2022-02-15:
Yes.
In standalone mode, there is some recommendation:
1. handle UX_DCD_ISR_PENDING, as background task (called by `ux_system_tasks_run`, which is polled in main loop), to handle control requests outside of interrupt services. Actually without this change the enumeration is fine, but there may be issue if non-control transfer could be interrupted by control requests.
2. Implement UX_DCD_TRANSFER_REQUEST in state machine way for _run functions (return _WAIT/_NEXT) for state progress, the parts may related is transfer implementation and ST HAL DCD callback implementation, since there could be status flags changes, the endpoint reset and abort process could also be changed.

AlexYzhov, 2022-02-15:
> Yes. In standalone mode, there is some recommendation:
>
> 1. handle UX_DCD_ISR_PENDING, as background task (called by `ux_system_tasks_run`, which is polled in main loop), to handle control requests [...]

Thanks for your suggestion, I'll do more research this weekend.

closed 2022-02-15 by AlexYzhov

AYEDMSTM, 2022-02-28:
HI @AlexYzhov ,

This is a sample example showing support of USB CDC-ACM Standalone mode  that can be accessed from ST github .  

https://github.com/STMicroelectronics/x-cube-azrtos-h7/tree/dev/usbx/Projects/NUCLEO-H723ZG/Applications/USBX/Ux_Device_CDC_ACM_Standalone

Regards,