eclipse-threadx/usbx #48: stm32 控制器中断服务函数中使用了 _ux_utility_memory_allocate 是否有内存隐患？

HelloByeAll, 2022-02-15:
stm32H7做 host 接了一个鼠标通过栈回溯了一次中断事件

```
-- OTG_FS_IRQHandler
   -> usbx\common\usbx_stm32_host_controllers\ux_hcd_stm32_interrupt_handler.c:81 (_ux_hcd_stm32_interrupt_handler)
      -> stm32h7xx_hal_hcd.c:579 (HAL_HCD_IRQHandler)
         -> stm32h7xx_hal_hcd.c:1238 (HCD_HC_IN_IRQHandler)
             -> usbx\common\usbx_stm32_host_controllers\ux_hcd_stm32_callback.c:304 (HAL_HCD_HC_NotifyURBChange_Callback)
                -> usbx\common\usbx_host_classes\src\ux_host_class_hid_transfer_request_completed.c:263
                   -> usbx\common\core\src\ux_host_stack_transfer_request.c:208
                      -> usbx\common\usbx_host_classes\src\ux_host_class_hid_transfer_request_completed.c:173
                         -> usbx\common\core\src\ux_utility_memory_allocate.c:199
                            -> \usbx\common\core\src\ux_utility_mutex_on.c:79
```

```
VOID  *_ux_utility_memory_allocate(ULONG memory_alignment, ULONG memory_cache_flag,
                                   ULONG memory_size_requested)
{

UX_MEMORY_BLOCK     *memory_block;
UX_MEMORY_BLOCK     *new_memory_block;
UX_MEMORY_BLOCK     *leftover_memory_block;
ULONG               memory_for_alignment;
ULONG               memory_removed_from_pool;
ULONG               leftover;
UCHAR               *memory_buffer;
ALIGN_TYPE          int_memory_buffer;

    /* Get the mutex as this is a critical section.  */
    _ux_utility_mutex_on(&_ux_system -> ux_system_mutex);

```

_txe_mutex_get中：
```
.......................

    /* Determine if everything is okay.  */
    if (status == TX_SUCCESS)
    {

        /* Check for interrupt call.  */
        if (TX_THREAD_GET_SYSTEM_STATE() != ((ULONG) 0))
        {
    
            /* Now, make sure the call is from an interrupt and not initialization.  */
            if (TX_THREAD_GET_SYSTEM_STATE() < TX_INITIALIZE_IN_PROGRESS)
            {
        
                /* Yes, invalid caller of this function, return appropriate error code.  */
                status =  TX_CALLER_ERROR;
            }
        }
    }

    /* Determine if everything is okay.  */
    if (status == TX_SUCCESS)
    {

        /* Call actual get mutex function.  */
        status =  _tx_mutex_get(mutex_ptr, wait_option);
    }

    /* Return completion status.  */
    return(status);
```
可以看到在中断中触发了一次 ` ux_utility_memory_allocate `，这个 ` mutex `由于是在中断中， 所以必然会返回 ` TX_CALLER_ERROR `，而调用处并没有对于` _ux_utility_mutex_on `的结果有任何判断，结尾处的 ` _ux_utility_mutex_off ` 同样如此，这个做法对于内存而言是否有安全隐患？

xiaocq2001, 2022-02-16:
Thanks for the feedback.
Yes, with RTOS in ISR if mutex_get is invoked there is potential issue.

The improve for HCD or DCD implement is, to move current interrupt callback handlings to a thread, then inside thread you can freely using RTOS features.

HelloByeAll, 2022-02-16:
> 

了解

closed 2022-02-18 by HelloByeAll