eclipse-threadx/usbx #50: wrong "packet_size" calculation for ISO/INT endpoint

alambe94, 2022-02-17:
Wrong calculation if wMaxPacketSize has multi transfer per microframe

In file "ux_host_stack_new_endpoint_create.c" line 157
`
        /* Number transaction over 2 additional is not allowed.  */
        n_tran = endpoint -> ux_endpoint_descriptor.wMaxPacketSize & UX_MAX_NUMBER_OF_TRANSACTIONS_MASK;
        if (n_tran >= UX_MAX_NUMBER_OF_TRANSACTIONS_MASK)
        {
            _ux_utility_memory_free(endpoint);
            return(UX_DESCRIPTOR_CORRUPTED);
        }
`
and at line no 
`
        /* Save final packet size.  */
        packet_size *= (n_tran + 1);
`
It should be 
`
        /* Number transaction over 2 additional is not allowed.  */
        n_tran = ((endpoint -> ux_endpoint_descriptor.wMaxPacketSize & UX_MAX_NUMBER_OF_TRANSACTIONS_MASK) >>
                UX_MAX_NUMBER_OF_TRANSACTIONS_SHIFT);

        if (n_tran > 2)
        {
            _ux_utility_memory_free(endpoint);
            return(UX_DESCRIPTOR_CORRUPTED);
        }
`

xiaocq2001, 2022-02-18:
Yes.
Before uses n_trans, there should be
`n_tran >>= UX_MAX_NUMBER_OF_TRANSACTIONS_SHIFT;`

I'm using
```
        n_tran = endpoint -> ux_endpoint_descriptor.wMaxPacketSize & UX_MAX_NUMBER_OF_TRANSACTIONS_MASK;
        if (n_tran >= UX_MAX_NUMBER_OF_TRANSACTIONS_MASK)
        {
            _ux_utility_memory_free(endpoint);
            return(UX_DESCRIPTOR_CORRUPTED);
        }
        n_tran >>= UX_MAX_NUMBER_OF_TRANSACTIONS_SHIFT;
```
to skip bit operation on error case.

alambe94, 2022-04-30:
Fixed in V6.1.11

closed 2022-04-30 by alambe94