eclipse-threadx/usbx #51: _ux_hcd_ohci_asynchronous_endpoint_create 中 ‘head_ed’ 需要用虚拟地址去访问 ux_ohci_ed_previous_ed

HelloByeAll, 2022-02-18:
```
head_ed =  (UX_OHCI_ED *) _ux_hcd_ohci_register_read(hcd_ohci, OHCI_HC_BULK_HEAD_ED);

```

```
    if (head_ed != UX_NULL)
        head_ed -> ux_ohci_ed_previous_ed =  ed;
```
head_ed 被直接从寄存器中读出的是物理地址，如果地址不是 1 : 1 映射关系，这个操作将会错误访问地址

修改为：
```
  if (head_ed != UX_NULL)
    {
        head_ed = _ux_utility_virtual_address(head_ed);
        head_ed->ux_ohci_ed_previous_ed = ed;
    }
```

HelloByeAll, 2022-02-19:
同样的问题在  ` _ux_hcd_ohci_asynchronous_endpoint_destroy.c line:167 ` 处也需要修改
修改前：
```
if (next_ed != UX_NULL)
        next_ed -> ux_ohci_ed_previous_ed =  previous_ed;
```

修改后：
```
next_ed = (ULONG)_ux_utility_virtual_address(next_ed);
if (next_ed != UX_NULL)
        next_ed -> ux_ohci_ed_previous_ed =  previous_ed;
```

xiaocq2001, 2022-02-21:
Thanks for the feedback.
That will be fixed.

HelloByeAll, 2022-05-10:
> Thanks for the feedback. That will be fixed.

最新版本 _ux_hcd_ohci_asynchronous_endpoint_create 中 head_ed  作为物理地址访问 ux_ohci_ed_previous_ed 是非法的。
此处应该为 ` head_ed = _ux_utility_virtual_address(head_ed); ` 
```
    /* Build the back chaining pointer. The previous head ED needs to know about the
       inserted ED. */
    if (head_ed != UX_NULL)
    {
        head_ed = _ux_utility_physical_address(head_ed);
        head_ed -> ux_ohci_ed_previous_ed =  ed;
    }
```

xiaocq2001, 2022-05-11:
Thanks. We will check all similar things and fix them.

xiaocq2001, 2023-01-30:
Fixed.

closed 2023-01-30 by xiaocq2001