eclipse-threadx/usbx #55: 怎么获取USB设备的生产商？

LiuChenghu, 2022-03-02:
获得了imanufacturer、iproduct、iserialnumber，如何通过这些索引获取字符串？

LiuChenghu, 2022-03-02:
还想请教一个问题，就是我使用USBX，识别到U盘后，过一会就一直进HAL_HCD_IRQHandler里面，其他程序无法执行，该如何排查问题？

xiaocq2001, 2022-03-03:
```
/*    device                                Pointer to device instance    */
/*    descriptor_buffer                     Pointer to a buffer to fill   */
/*                                          LANGID or STRING descriptor   */
/*    length                                Length of buffer              */
/*    language_id                           0 to obtain LANGID descriptor */
/*                                          valid language ID to obtain   */
/*                                          string descriptor             */
/*    string_index                          Index of the string           */
```
For returned descriptor, please refer to section 9.6.7 String of USB 2.0 spec.

xiaocq2001, 2022-03-03:
If your program keep interrupted and runs in HAL_HCD_IRQHandler, please check if some interrupt is not correctly handled so it keeps triggering interrupt.

xiaocq2001, 2022-08-11:
No feedback long time, closing.

closed 2022-08-11 by xiaocq2001