eclipse-threadx/usbx #58: Provide sample code for usb audio device class

jbruneaux31, 2022-03-14:
Hi,
Can you please provide sample code for USB audio device class ?
I've seen the partial example here :
https://docs.microsoft.com/fr-fr/azure/rtos/usbx/usbx-device-stack-supplemental-2#usb-device-audio-class

But there's still a lot of missing bits to have a functional example. The API documentation is quite short.

Regards,

xiaocq2001, 2022-03-22:
Thanks for the feedback.
We will share you when there is example project.

jbruneaux31, 2022-03-22:
Hi,

When looking at the issue here : https://github.com/azure-rtos/usbx/issues/9 , it looked like there were already a demo code available. Even if the code is not complete, maybe the template can be shared ?

Regards,

xiaocq2001, 2022-03-23:
Sure, you can use the template.

jbruneaux31, 2022-03-23:
Hi,

I meant that there were no template availbale, only some part of it. Some defines used in the official documentation are only provided in the issue #9 but there's nothing that we can use as a template to at least have the usbx usage part.

The problem is that using the codes snippets from the demo example, I can get the USB audio device enumerated on the computer and I see that there are some URB transfers to the device (using wireshark). Unfortunately, the ux_device_class_audio_stream_frame_done callback is never called. 
ux_device_class_audio_stream_change is called, ux_slave_class_audio_instance_activate and ux_slave_class_audio_instance_deactivate as well as ux_device_class_audio_control_process callbacks are called and I can see the volume change requests. 
The problem is that I don't 'see' the received audio frames.

There's a lack of documentation on the expected behavior of the audio class (what is the usage of the callbacks). 

Regards,

xiaocq2001, 2022-03-23:
Audio support needs isochronous transfer support (currently needed for UAC and UVC, but not other classes), so if device controller driver is not ready for this transfer mode, the transfer done callback will never happen.

The instance activate/deactivate/change are actually related to control transfer which is the default supported transfer mode for all devices and classes.

So the situation could be: your DCD driver is ready for device enumeration (control transfer ready) but there is still issue to be solved to let isochronous transfer run.

jbruneaux31, 2022-03-23:
Hi,

Thanks for this clarification. I'll check the DCD driver. 

Regards

jbruneaux31, 2022-03-23:
Hi,

DCD driver (stm32) to allow isochronous transfers, but after playing around with the test project I'm working on, I've only been able to see some transfers (not working as expected thus) after calling ux_device_class_audio_reception_start, but I don't know if this is the normal usage. 
Do I have to call any usbx function (like the ux_device_class_audio_reception_start function) during one of the callbacks ? 

Regards,

xiaocq2001, 2022-03-24:
No, that's called once before writing frames.
Maybe the STM32 HAL version has some issue on ISO transfer support. In ST device controller there is iso incomplete design, please check HAL and DCD to confirm if that's handled.

jbruneaux31, 2022-03-24:
The ux_device_class_audio_reception_start function is not called anywhere in the usbx code. Reading the code of this functions shows that it's not related to writing frame (I'm trying to receive audio frame from the host).
The DCD do not handle the ISO incomplete notification. During the last tests yesterday, I've added the ISO incomplete callback. I've seen that the ISO Incomplete callback of the HAL driver is called (a lot of time) but only after the call to "ux_device_class_audio_reception_start" (which probably start a transfer). The problem is that I don't know what to do in the ISO incomplete callback. Do I have to pass something to the usbx stack ?

Is there any DCD driver for an other platform which work with isochronous transfers (in the examples projects), so that I can check how it should be handled.

Regards, Jerome

yuxin-azrtos, 2022-04-01:
@jbruneaux31 which chip/eval-board do you use?

jbruneaux31, 2022-04-01:
Hi,

I'm using the STM32F429ZI on the NUCLEO-F429ZI evaluation board.

Regards

AYEDMSTM, 2022-04-07:
Hi, 

We are working to provide USBX device audio application running on STM32H7 platform as a reference application that can be ported to STM32F4.

Regards ST Team.

fyyxxm, 2022-04-11:
> Hi,
> 
> We are working to provide USBX device audio application running on STM32H7 platform as a reference application that can be ported to STM32F4.
> 
> Regards ST Team.

sounds great!

closed 2022-04-26 by yuxin-azrtos

AYEDMSTM, 2022-05-09:
Hi @jbruneaux31,

This is a sample example showing support of Audio Class that can be accessed from ST github .

https://github.com/STMicroelectronics/x-cube-azrtos-h7/tree/dev/usbx/Projects/STM32H743I-EVAL/Applications/USBX/Ux_Device_Audio

Regards ST Team.