eclipse-threadx/usbx #62: ux_network_driver

chengshuihang, 2022-04-23:
我想支持下NX_LINK_GET_STATUS，下面代码可以吗？
[image]
我发现关于ux_network_device_link_status的操作都是没有mutex的，在USB网卡插拔或者复位的时候不会对Netxduo造成破坏性影响吗？

另外ux_network_driver好像不支持IPV6，而且IPV6宏有个小错误
[image]

xiaocq2001, 2022-04-24:
For IPV6 support, `NX_ETHERNET_IPV6` should be 0x86DD (ref [here](https://github.com/azure-rtos/netxduo/blob/1eee883def078a91b8cd19159d1796e3a8862814/common/src/nx_ram_network_driver.c#L55)),
To support IPV6 incoming packet, additional change in `_ux_network_driver_packet_received` should be made to accept IPV6 packets (ref [here](https://github.com/azure-rtos/netxduo/blob/1eee883def078a91b8cd19159d1796e3a8862814/common/src/nx_ram_network_driver.c#L883)).

chengshuihang, 2022-04-24:
是的，我修改了_ux_network_driver_packet_received和_ux_host_class_cdc_ecm_thread函数，让它们可以处理IPV6数据包。

xiaocq2001, 2022-08-11:
network driver updated with IPV6 and some other fixes.

closed 2023-01-30 by xiaocq2001