eclipse-threadx/usbx #63: USBX drives CDC-ECM device as USB host

rk3399, 2022-05-04:
When the CDC-ECM device reports the MAC address to USBX, the whole system stops. I would like to know what is the reason for this.
My guess is that USBX is waiting for the CDC-ECM device to report the network connection status. But the CDC-ECM device doesn't do that. If my guess is correct, can you tell me where to find documentation on the CDC-ECM protocol, thanks!
(From Google Translate, the following is the original Chinese text.)

当CDC-ECM设备向USBX报告MAC地址后，整个系统便停止了。我想知道这是什么原因。
我的猜测是USBX在等待CDC-ECM设备报告网络连接状态。但CDC-ECM设备并没有这样做。如果我的猜测是对的，可以告诉我在哪里能找到关于CDC-ECM协议的文档吗，感谢！

rk3399, 2022-05-04:
何时会从这个等待返回
[image]

rk3399, 2022-05-04:
何时会从这个等待返回
[image]

xiaocq2001, 2022-05-05:
See ux_host_class_cdc_ecm_interrupt_notification.c, the semaphore is put when there is interrupt transfer notification about link state changes.

xiaocq2001, 2022-05-05:
Please check [this file](https://usb.org/sites/default/files/CDC1.2_WMC1.1_012011.zip) for CDC ECM spec.

rk3399, 2022-05-05:
> Please check [CDC1.2_WMC1.1_012011.zip on USB.org](https://usb.org/sites/default/files/CDC1.2_WMC1.1_012011.zip) for CDC ECM spec.
Thanks, this solved my doubt!

closed 2022-05-05 by rk3399