eclipse-threadx/usbx #66: Buffer overflow in ux_device_class_cdc_acm_read

KabaTrinker, 2022-07-18:
I do an call to ux_device_class_cdc_acm_read with an buffer with size 150 and a requested_length=150 for example. If I send now 160 byte, the function returns with actual_length=160 and my buffer was filled with 160 bytes, so I've an overflow.
Is this behavior correct?

xiaocq2001, 2022-07-19:
Yes. You may overflow. It seems in DCD transfer implement, the actual transferred size is directly reported without checking if received number of bytes exceeds requested.

For reading, it's recommended to offer full packet buffer (n*64 for FS and n*512 for HS), since USB data is transferred packet by packet, and short packet (data less than a full packet) terminates the transfer.

KabaTrinker, 2022-07-20:
Ok, if I use a buffer size of n * packet size then I've no buffer overflow.
Thank you for your fast response.

closed 2022-07-20 by KabaTrinker