eclipse-threadx/usbx #71: OHCI 插入设备上电不识别，POTPGT延时可能存在问题

HelloByeAll, 2022-08-15:
我的ohci设备在插入设备之后再上电枚举执行到 Set Address时没有接收0长度响应包， 此时拔插设备后可以正常枚举。
我搜索到相关问题，和我的情况有些类似，虽然没有解决问题，不过发现usbx POTPGT 延时确实可能存在一些问题。

如果是插入设备再上电的情况， 那么port 电源使能之后会立刻触发 ` OHCI_HC_INT_RHSC `中断，
` _ux_utility_delay_ms(ohci_register_a >> (OHCI_HC_RH_POTPGT - 1)); ` 延时将失去作用

[image]

类似问题：
https://www.cnblogs.com/sankye/articles/1584487.html

不响应0长度响应包：
[image]

正常枚举抓包：
[image]

xiaocq2001, 2022-08-15:
Hi, did you try to disable RHSC and enable it after power on?

HelloByeAll, 2022-08-15:
> ，您是否尝试禁止 RHSC 并在启动后您好？

是的， 我尝试在ux_hcd_ohci_initialize中屏蔽了RHSC中断使能， 并在初始化完成后重新使能RHSC中断， 现象依然存在
[image]

HelloByeAll, 2022-08-16:
> Hi, did you try to disable RHSC and enable it after power on?

通过逻辑分析仪抓包发现是设备没有发送， 并不是没有接收到

closed 2022-08-16 by HelloByeAll