eclipse-threadx/usbx #72: Any guide/doc on how to port to new hardware?

josesimoes, 2022-08-16:
I'm looking at using USBx in a not-currently-supported-platform. I've already found #10 and #12. That's perfectly clear that the integration has to be done by silicon vendors or others need those.

What I can not find is documentation or guidance on how to go through that port process. I understand that I can start looking into one of the available ports from one of the vendors, but that doesn't look very effective.

Any pointers and guidance on this is much appreciated!

goldscott, 2022-08-16:
We do not have any sort of porting guide for USBX. Your best bet is to look at a couple of the available ports. What platform are you looking at?

josesimoes, 2022-08-16:
Platform is Silabs Giant Gecko S1.

That's really unfortunate!! 😯 

I'll have to understand two platforms: the one that I'll be getting inspiration from and the target one. Not very efficient...

May I suggest that such documentation is produced? Kind of what exists for Azure RTOS NetX Duo Network Drivers. Otherwise that can be _painfull_ and not very efficient to tackle.