eclipse-threadx/usbx #74: Where can we download the USBX OTG sample code based the NXP LPC3131?

Barfell, 2022-09-17:
I found the USBX OTG charpter on microsoft docs, the sample code based NXP LPC 3131, how can we down load the sample project code?

Interrupt handlers
Both host and device controller drivers for OTG needs different interrupt handlers to monitor signals beyond traditional USB interrupts, in particular signals due to SRP and VBUS.

How to initialize a USB OTG controller. We use the NXP LPC3131 as an example here.

/* Initialize the LPC3131 OTG controller. */
status = ux_otg_lpc3131_initialize(0x19000000, lpc3131_vbus_function,
    tx_demo_change_mode_callback);
In this example, we initialize the LPC3131 in OTG mode by passing a VBUS function and a callback for mode change (from host to slave or vice versa).

The callback function should simply record the new mode and wake up a pending thread to act up the new state.

xiaocq2001, 2022-09-23:
Maybe you can check if that is in NXP LPC3131 SDK?

Barfell, 2022-09-23:
> 

Not found NXP LPC3131 SDK in nxp.com.

xiaocq2001, 2022-09-27:
Unfortunately the LPC3131 is quite old and there is no live example project right now.

Barfell, 2022-09-27:
In order to promote THREADX, Microsoft has specially produced a sample document, which is an example made of LPC3131 in the document, and does not provide a  DEMO project.