eclipse-threadx/usbx #78: Wrong CDC class ID

josesimoes, 2023-01-13:
**Describe the bug**

I'm trying to setup a STM32F769 as USB device implementing a Virtual COM port with USBX.
I'm having issues on the device as it doesn't perform as expected right after the enumeration is complete.
I've traced down the problem to the CDC class id being wrong.

According to [USB docs](https://www.usb.org/defined-class-codes) the class ID should be 0x02.
In USBX this is defined as 10 [here](https://github.com/azure-rtos/usbx/blob/2c66db9568b0adbae6b06a698d9888b664d02379/common/usbx_device_classes/inc/ux_device_class_cdc_acm.h#L82) which seems to be wrong.
As soon as I change it to 0x02, the check [here](https://github.com/azure-rtos/usbx/blob/2c66db9568b0adbae6b06a698d9888b664d02379/common/usbx_device_classes/src/ux_device_class_cdc_acm_entry.c#L108) is successful and the code continues happily. 

**To Reproduce**

**Expected behavior**

**Impact**
Prevents the application from working.

**Logs and console output**
N/A

**Additional context**
N/A

xiaocq2001, 2023-01-28:
Thanks for the feedback.
Are you using CDC-ACM class with IAD or without IAD?
If your framework descriptor does not include IAD can you try to add IAD and check if it works with the previous definition? Thanks.

josesimoes, 2023-01-30:
@xiaocq2001 thanks for looking into this.

I've tried a descriptor with and without IAD. The outcome was the same...

xiaocq2001, 2023-01-31:
CDC-ACM combines two interfaces, the first interface is control and class ID is 2, the second one is data and class ID is 10, in USBX device stack the IAD is checked to associate class driver to both CDC-ACM interfaces, the data interface is checked to obtain endpoints to communicate. This have been working for long time.

As far as I know, ST provided USBX device CDC-ACM example, can you check that as reference?

josesimoes, 2023-02-24:
Yes, this is the code I've been using for reference: https://github.com/STMicroelectronics/x-cube-azrtos-f7/tree/main/Projects/STM32F769I-Discovery/Applications/USBX/Ux_Device_CDC_ACM.

I can confirm that I can run the project successfully on their STM32F769I-Discovery from the STM32 Cube IDE.
The reason for mentioning this is that my project is related with .NET nanoFramework: I'm adding support for USBX in our official Azure RTOS distribution, for the STM32 platform. 

The board I'm using it's a custom one, based on the same STM32F769I part, just that it doesn't have the USB PHY as the STM32F769I-Discovery has. Could it be that it can explain the "misbehave" I'm seeing here?

xiaocq2001, 2023-02-27:
From your description, it seems the same code (original example) has different behavior on different board. It's better to get actual flow captured to see what's the difference to analyze.