eclipse-threadx/usbx #80: USBX support on STM32F7 with two separate USB interfaces

sw2mrc, 2023-01-24:
Good day!

Board I am using is STM32F746-DISCO, the latest F7 firmware pkg from ST is being used, and the version of USBX packaged is 6.1.10. 

F746 mcu has two usb interfaces, and I'm trying to use one interface as usb-host and the other as usb-device simultaneously, and this causes only one usb function to work, not both; however the firmware does not crash. Also, usbx examples that use just one usb interface for host-only or device-only functionality work fine.

It appears that when both device and host functionality are enabled simultaneously, the most recently initialized function works. For example, when usb host init is done after usb device init, then only host functionality works (and vice versa).

Separate byte pools of size 70kb and 7kb are allocated to host and device,  and while the pools being passed are separate, at the USBX core doesn't seem to maintain interface separation from a cursory look the function ux_system_initialize in file ./Middlewares/ST/usbx/common/core/src/ux_system_initialize.c

This leads me to believe that using two interfaces simultaneously as two usb devices, or two usb hosts or a mix would not work either.

I had to comment out the defines UX_HOST_SIDE_ONLY, and UX_DEVICE_SIDE_ONLY in generated file USBX/App/ux_user.h to make the example not crash, but these defs seem to be for OTG functionality of a specific usb interface, which I'm not using.

Any workarounds that can be used to make progress would be highly appreciated.

STM32F7/F4 and many other MCUs support two usb interfaces and USBX supporting their concurrent use would be very essential.

Thank you.

xiaocq2001, 2023-01-28:
Thanks for using USBX.
I assume you are running host and device on the different USB peripheral (different interrupt and register base), in this case, there should be no conflict for the peripherals to work at the same time. In application there are several things to check:

1. the IRQ handler should be set up to invoke DCD/HCD handler correctly.
2. `ux_system_initialize` must be called ONLY once, to manage memory for both host and device.
3. the other device side initialization hand host side initialization sequence should be the same as what you did as device role only or host side role only.
4. the HCD and DCD initialization should be called with the correct (different) register base to start up.

BluesharkPD, 2023-02-01:
Hi, 

It's possible to run 2 cdc host class with two USB port in the same time ??

xiaocq2001, 2023-02-02:
Yes.
1. If two port is on same controller's root hub, just register the HCD.
2. If two port is on different controller, just register the HCD with different initialization parameter to bring them up.

sw2mrc, 2023-02-02:
Just wanted to confirm the original issue I reported was resolved when ux_system_initialize was called just once. Thanks much you for your help! Cheers.

closed 2023-02-02 by sw2mrc