eclipse-threadx/usbx #81: HID RCU multi report

BluesharkPD, 2023-02-07:
Hi,
i have an hid device with multi hid report ID , it seems that USBX HID RCU stack parser only the first Report id.

BluesharkPD, 2023-02-08:
This is HID Report Descriptor of my headset device

ID Descriptor
Usage Page: 11 (Telephony)
Usage: 5 (Headset)
Collection 
Report ID: 2
Usage Page: 11 (Telephony)
Logical Minimum: 0
Logical Maximum: 1
Usage: 32 (Hook Switch)
Usage: 151 (Line Busy Tone)
Report Size: 1
Report Count: 2
Input: 34 (Data,Variable,Absolute,No Wrap,Linear,No Preferred State,No Null Position)

Usage: 47 (Phone Mute)
Usage: 33 (Flash)
Usage: 36 (Redial)
Usage: 112 (Voice Mail)
Report Size: 1
Report Count: 4
Input: 7 (Constant,Variable,Relative,No Wrap,Linear,Preferred State,No Null Position)

Usage: 7 (Programmable Button)
Usage Page: 9 (Buttons)
Report Size: 1
Report Count: 1
Input: 2 (Data,Variable,Absolute,No Wrap,Linear,Preferred State,No Null Position)

Report Size: 1
Report Count: 9
Input: 1 (Constant,Array,Absolute,No Wrap,Linear,Preferred State,No Null Position)

Usage Page: 8 (LEDs)
Logical Minimum: 0
Logical Maximum: 1
Usage: 23 (Off-Hook)
Usage: 9 (Mute)
Usage: 24 (Ring)
Usage: 32 (Hold)
Usage: 33 (Microphone)
Report Size: 1
Report Count: 5
Output: 34 (Data,Variable,Absolute,No Wrap,Linear,No Preferred State,No Null Position,Non Volatile)

Usage Page: 11 (Telephony)
Logical Minimum: 0
Logical Maximum: 1
Usage: 158 (Unknown)
Report Size: 1
Report Count: 1
Output: 34 (Data,Variable,Absolute,No Wrap,Linear,No Preferred State,No Null Position,Non Volatile)

Report Size: 1
Report Count: 10
Output: 1 (Constant,Array,Absolute,No Wrap,Linear,Preferred State,No Null Position,Non Volatile)

End Collection 
Usage Page: 65280 (Vendor Specific)
Usage: 5 (Unknown)
Collection 
Report ID: 4
Usage Page: 65328 (Vendor Specific)
Logical Minimum: 0
Logical Maximum: 1
Usage: 65535 (Unknown)
Report Size: 1
Report Count: 1
Feature: 34 (Data,Variable,Absolute,No Wrap,Linear,No Preferred State,No Null Position)

Report Size: 1
Report Count: 7
Feature: 1 (Constant,Array,Absolute,No Wrap,Linear,Preferred State,No Null Position)

Usage: 32 (Unknown)
Usage: 151 (Unknown)
Report Size: 1
Report Count: 2
Input: 34 (Data,Variable,Absolute,No Wrap,Linear,No Preferred State,No Null Position)

Usage: 47 (Unknown)
Usage: 33 (Unknown)
Usage: 36 (Unknown)
Usage: 65533 (Unknown)
Report Size: 1
Report Count: 4
Input: 7 (Constant,Variable,Relative,No Wrap,Linear,Preferred State,No Null Position)

Report Size: 1
Report Count: 10
Input: 1 (Constant,Array,Absolute,No Wrap,Linear,Preferred State,No Null Position)

Usage Page: 65344 (Vendor Specific)
Logical Minimum: 0
Logical Maximum: 1
Usage: 23 (Unknown)
Usage: 9 (Unknown)
Usage: 24 (Unknown)
Usage: 32 (Unknown)
Usage: 33 (Unknown)
Report Size: 1
Report Count: 5
Output: 34 (Data,Variable,Absolute,No Wrap,Linear,No Preferred State,No Null Position,Non Volatile)

Usage Page: 65328 (Vendor Specific)
Logical Minimum: 0
Logical Maximum: 1
Usage: 158 (Unknown)
Report Size: 1
Report Count: 1
Output: 34 (Data,Variable,Absolute,No Wrap,Linear,No Preferred State,No Null Position,Non Volatile)

Report Size: 1
Report Count: 10
Output: 1 (Constant,Array,Absolute,No Wrap,Linear,Preferred State,No Null Position,Non Volatile)

Report ID: 5
Usage Page: 65280 (Vendor Specific)
Usage: 1 (Unknown)
Logical Minimum: 0
Logical Maximum: 255
Report Size: 8
Report Count: 32
Output: 258 (Data,Variable,Absolute,No Wrap,Linear,Preferred State,No Null Position,Non Volatile)

Usage: 1 (Unknown)
Logical Minimum: 0
Logical Maximum: 255
Report Size: 8
Report Count: 32
Input: 258 (Data,Variable,Absolute,No Wrap,Linear,Preferred State,No Null Position)

End Collection 
Usage Page: 12 (Consumer)
Usage: 1 (Consumer Control)
Collection 
Report ID: 1
Usage Page: 12 (Consumer)
Logical Minimum: 0
Logical Maximum: 1
Usage: 234 (Volume Decrement)
Usage: 233 (Volume Increment)
Usage: 263 (Fire Alarm)
Report Size: 1
Report Count: 3
Input: 2 (Data,Variable,Absolute,No Wrap,Linear,Preferred State,No Null Position)

Report Size: 1
Report Count: 13
Input: 1 (Constant,Array,Absolute,No Wrap,Linear,Preferred State,No Null Position)

End Collection

xiaocq2001, 2023-02-10:
On host HID side, the clients supported are keyboard, mouse and remote control. The other page/usage is not supported right now.

You can register callback through `ux_host_class_hid_report_callback_register `with flag `UX_HOST_CLASS_HID_REPORT_RAW `to handle raw report data to extract things by yourself.

BluesharkPD, 2023-02-10:
In the HID Report remote control is here  but it is in the end.
[image]

xiaocq2001, 2023-02-13:
Yes. in ux_host_class_hid_transfer_request_completed.c only one report is processed.

We will discuss and fix. Your suggestion on fixing is also appreciated.

Thanks.