eclipse-threadx/usbx #82: NXP MIMXRT1170 SRAM device storage

BluesharkPD, 2023-02-13:
Hi, I would like to develop an application device storage SRAM without RTOS. using a NXP MIMXRT1170. if i include usbx and filex there is warning of duplicated declaration  of Var in port. and declaration of var are not similar in tx/fx/ux port.

for example 
ULONG64 declared in fx 
typedef unsigned long long                      ULONG64;

ULONG64 declared in fx 
typedef unsigned long long                      ULONG64;

ULONG64 declared in ux
typedef uint64_t                                ULONG64;

bhnaphade, 2023-02-13:
@BluesharkPD thank you for reporting the warning. we will check this internally and address this soon.

BluesharkPD, 2023-03-02:
Any update about this (patch or expand to see details).??

closed 2023-05-18 by BluesharkPD