eclipse-threadx/usbx #83: HID, Device Set Feature Report (wrong Len)

sebastian-dev, 2023-02-15:
File: ux_device_class_hid_report_set.c, function: _ux_device_class_hid_report_set
I have a setup with an USBX HID as device.
I Use Feature Reports.
The Reported Data size is always: UX_DEVICE_CLASS_HID_EVENT_BUFFER_LENGTH
In my case i have a report size of 101, Endpoint size 64

When the device receive a feature Report, the fuction: _ux_device_class_hid_report_set
is called before the callback "hid -> ux_device_class_hid_callback" is called.

The reported size in the result is wrong. ux_slave_transfer_request_actual_length is always 0.
This looks wrong:
`id_event.ux_device_class_hid_event_length = transfer_request -> ux_slave_transfer_request_actual_length -1;`
this:
`hid_event.ux_device_class_hid_event_length = host_length - 1;` make more sense.

ux_slave_transfer_request_actual_length  is the actual current amount of data in the endpoint, not the total transfered data len.

xiaocq2001, 2023-02-16:
Feature request is sent through control request, the host_length is from wLength in SETUP packet, but host may still issue short packet to make data shorter than the wLength.

So check _actual_length which is actual data received from host after the SETUP should be better.

I think the issue here is in DCD, where in control endpoint interrupt service the length of actual data received is not correctly written to _actual_length.

sebastian-dev, 2023-02-16:
the property ux_slave_transfer_request_actual_length is only set in the host class: 

`
/* Add to the actual payload length.  */
slave_transfer_request -> ux_slave_transfer_request_actual_length += td_length;
`

but never in the the device class: 
`
/* Reset the number of bytes sent/received.  */
        transfer_request -> ux_slave_transfer_request_actual_length =  0;
`

when you search over the complete usbx stack: ux_slave_transfer_request_actual_length is never set in a device class to an other value than 0.
THe value is evaluated in several device classes, but never set.

sebastian-dev, 2023-02-16:
Looks like maybe an issue with the Renesas USBX low level implementation.
During the control transfer no sizes updated in &device->ux_slave_device_control_endpoint.ux_slave_endpoint_transfer_request;