eclipse-threadx/usbx #86: Compiler warnings in GCC 12.2.rel1

josesimoes, 2023-02-24:
**Describe the bug**
When compiling with GCC 12.2.rel1 and strict compiler warnings, the build fails with these

```text
[build] E:/GitHub/nf-interpreter/build/_deps/azure_rtos_usbx-src/ports/cortex_m7/gnu/inc/ux_port.h:214:60: error: array subscript 0 is outside array bounds of 'ULONG[0]' {aka 'long unsigned int[]'} [-Werror=array-bounds]
[build]   214 | #define outpl(a, b)                                        *((ULONG *)  (a)) =  ((ULONG)  (b))
[build]       |                                                            ^~~~~~~~~~~~~~~~~
[build] E:/GitHub/nf-interpreter/build/_deps/azure_rtos_usbx-src/common/core/src/ux_utility_pci_write.c:100:9: note: in expansion of macro 'outpl'
[build]   100 |         outpl(UX_PCI_CFG_CTRL_ADDRESS, cfg_ctrl);
[build]       |         ^~~~~
[build] E:/GitHub/nf-interpreter/build/_deps/azure_rtos_usbx-src/ports/cortex_m7/gnu/inc/ux_port.h:214:60: error: array subscript 0 is outside array bounds of 'ULONG[0]' {aka 'long unsigned int[]'} [-Werror=array-bounds]
[build]   214 | #define outpl(a, b)                                        *((ULONG *)  (a)) =  ((ULONG)  (b))
[build]       |                                                            ^~~~~~~~~~~~~~~~~
[build] E:/GitHub/nf-interpreter/build/_deps/azure_rtos_usbx-src/common/core/src/ux_utility_pci_write.c:103:9: note: in expansion of macro 'outpl'
[build]   103 |         outpl(UX_PCI_CFG_DATA_ADDRESS, value);
[build]       |         ^~~~~
[build] E:/GitHub/nf-interpreter/build/_deps/azure_rtos_usbx-src/ports/cortex_m7/gnu/inc/ux_port.h:214:60: error: array subscript 0 is outside array bounds of 'ULONG[0]' {aka 'long unsigned int[]'} [-Werror=array-bounds]
[build]   214 | #define outpl(a, b)                                        *((ULONG *)  (a)) =  ((ULONG)  (b))
[build]       |                                                            ^~~~~~~~~~~~~~~~~
[build] E:/GitHub/nf-interpreter/build/_deps/azure_rtos_usbx-src/common/core/src/ux_utility_pci_write.c:109:9: note: in expansion of macro 'outpl'
[build]   109 |         outpl(UX_PCI_CFG_CTRL_ADDRESS, cfg_ctrl);
[build]       |         ^~~~~
[build] E:/GitHub/nf-interpreter/build/_deps/azure_rtos_usbx-src/ports/cortex_m7/gnu/inc/ux_port.h:214:60: error: array subscript 0 is outside array bounds of 'ULONG[0]' {aka 'long unsigned int[]'} [-Werror=array-bounds]
[build]   214 | #define outpl(a, b)                                        *((ULONG *)  (a)) =  ((ULONG)  (b))
[build]       |                                                            ^~~~~~~~~~~~~~~~~
[build] E:/GitHub/nf-interpreter/build/_deps/azure_rtos_usbx-src/common/core/src/ux_utility_pci_write.c:118:9: note: in expansion of macro 'outpl'
[build]   118 |         outpl(UX_PCI_CFG_CTRL_ADDRESS, cfg_ctrl);
[build]       |         ^~~~~
```

xiaocq2001, 2023-02-24:
Hi, it seems like a bug since the code here just covert const to ULONG pointer to access it as ULONG, there is no errror generated in previous compilers.

I searched and find some bugs as follow:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=104657
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=99578

josesimoes, 2023-02-25:
OK. If you don't think this is relevant please just go ahead and close it.
I'll have to suppress this on our pipeline.

xiaocq2001, 2023-02-27:
OK. Closed.

closed 2023-02-27 by xiaocq2001