eclipse-threadx/usbx #87: I coudn't able to find uvc device class support sample application in usbx

Mahesh-dev-avula, 2023-03-02:
**Describe the bug**
Actually I am trying to give uvc device class support, I am looking for any sample application or any document to understand the flow of UVC.

Can anyone help on this. 

Please also mention any information which could help others to understand
the problem you're facing:
- I am using target board as cortex m55
- release version 6.2.0

Mahesh-dev-avula, 2023-03-03:
Hi @xiaocq2001 ,
Thank you so much for detailed information.

Mahesh-dev-avula, 2023-03-03:
Hi @xiaocq2001 ,

I have one doubt that, as you explained _max_payload_buffer_nb and _max_payload_buffer_size both the variables size would same as Isochrnous endpoint size..?
if I give both sizes as Isocronous endpoint size, class registration was failing because it requires almost 1MB memory, i have less memory on my device.
 **memory_size = stream -> ux_device_class_video_stream_payload_buffer_size *
                            stream_parameter -> ux_device_class_video_stream_parameter_max_payload_buffer_nb;**
Could you please confirm that.

Thanks
Mahesh Avula

xiaocq2001, 2023-03-06:
_buffer_size is max endpoint size, _buffer_nb is number of payloads buffered.

Mahesh-dev-avula, 2023-03-06:
Thank you so much  @xiaocq2001

Mahesh-dev-avula, 2023-03-07:
Hi @xiaocq2001 ,

Note: 1

Below are descriptors for video class driver which i have prepared, with these descriptors,
Isochrnous endpoint is not creating because of this endpoint is associated with video stream interface
with alternate setting  =1, I have gone through code and found that my device not getting UX_SET_INTERFACE
request from host machine(Ubuntu 18.04)

Could you please help on this issue, if i am doing anything wrong here.

UCHAR device_framework_high_speed[] = {

    /*
    Device descriptor 18 bytes
    0x00 bDeviceClass: base class, the class code will be defining in interface descriptor
    */
    0x12, 0x01, 0x00, 0x00,
    0xEF, 0x02, 0x01, 0x40,
    0x25, 0x05, 0xa7, 0xa4,
    0x00, 0x01, 0x01, 0x02,
    0x03, 0x01,

    /* Configuration  descriptor 9 bytes  */
    0x09, 0x02, 0xb5, 0x00, 0x02, 0x01, 0x00, 0xC0, 0x01,

    /* Interface association descriptor  */
    0x08, 0x0b, 0x00,
    0x02, 0x0E, 0x03, 0x00, 0x00,

    /* interface_descriptor UVC_control */
    0x09, 0x04, 0x00, 0x00, 0x01, 0x0E, 0x01, 0x00, 0x00,

    /* Class-specific VC Interface Header Descriptor */
     0x0d,0x24,0x01,0x10,0x01,0x34,0x00,0x00,0x00,0x00,0x00,0x01, 0x01,

      /*  Input Terminal (VC_INPUT_TERMINAL, Camera)  */
     0x11,0x24,0x02,0x01,0x01,0x02,0x00, 0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x02,0x00,0x00,

     /*  Output Terminal (VC_OUTPUT_TERMINAL, USB)  */
     0x09, 0x24, 0x03, 0x02,0x01,0x01,0x00,0x05,0x00,

      /* Processing Unit (VC_PROCESSING_UNIT)  */
     0x0d, 0x24, 0x05,0x05,0x01,0x00,0x00,0x03,0x00,0x00,0x00,0x00,0x00,

    /* interrupt endpoint_descriptor for uvc_control 7 bytes */
    0x07,0x05,0x83,0x03,0x08,0x00,0x08,

    /* Interface Descriptor for UVC streaming where altsetting = 0 */
    0x09, 0x04, 0x01, 0x00, 0x00, 0x0e, 0x02, 0x00, 0x00,

    /* Class Video streaming input Header Descriptor (VS_INPUT_HEADER)  */
    0x0e,0x24,0x01,0x01,0x47,00,0x81,0x00,0x02,0x00,0x00,0x00,0x01,0x00,

    /* VS Format Descriptor (VS_FORMAT_UNCOMPRESSED)  */
    0x1b,0x24,0x04,0x01,0x01,0x59,0x55,0x59,0x32, 0x00,0x00, 0x10,0x00, 0x80,0x00, 0x00,0xAA,0x00,0x38,0x9B,0x71,0x10,0x01,0x00,0x00,0x00,0x00,

     /* VS Frame Descriptor (VS_FRAME_UNCOMPRESSED)  */
    0x1e,0x24, 0x05,0x01, 0x00,0x00,0x1e,0x00,0x1e,0x00,0xec,0x0d,0x00,0x00,0xec,0x0d,0x00,0x00,0x8c,0x0a,0x00,0x00,0x64,0x00,0x00,0x01,0x00,0x64,0x00,0x00,

     /* Interface Descriptor for UVC streaming where altsetting = 1 */
    0x09, 0x04, 0x01, 0x01, 0x01, 0x0e, 0x02, 0x00, 0x00,

    /* Isochronous IN endpoint descriptor for UVC streaming  */
    0x07, 0x05, 0x85, 0x05, 0x00, 0x04, 0x01,
	
}

Note: 2

When I run my application I have collected dmesg logs at host side(ubuntu18.04) and attaching here.
 node has not created for video class driver in /dev/ directory.

[18159350.110416] usb 1-4.4.4.2: new high-speed USB device number 78 using xhci_hcd
[18159350.210765] usb 1-4.4.4.2: config 1 interface 1 altsetting 1 endpoint 0x85 has an invalid bInterval 0, changing to 7
[18159350.211131] usb 1-4.4.4.2: New USB device found, idVendor=0525, idProduct=a4a7
[18159350.211132] usb 1-4.4.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[18159350.211133] usb 1-4.4.4.2: Product: Devkit
[18159350.211134] usb 1-4.4.4.2: Manufacturer: AlifSemiconductor
[18159350.211152] usb 1-4.4.4.2: SerialNumber: 1200
[18159350.274379] uvcvideo: Found UVC 1.10 device Devkit (0525:a4a7)
[18159360.538136] uvcvideo: UVC non compliance - GET_DEF(PROBE) not supported. Enabling workaround.
[18159365.657965] uvcvideo: Failed to query (129) UVC probe control : -110 (exp. 34).
[18159365.657966] uvcvideo: Failed to initialize the device (-5).

Regards,
Mahesh

xiaocq2001, 2023-03-08:
> [18159360.538136] uvcvideo: UVC non compliance - GET_DEF(PROBE) not supported. Enabling workaround.
> [18159365.657965] uvcvideo: Failed to query (129) UVC probe control : -110 (exp. 34).
> [18159365.657966] uvcvideo: Failed to initialize the device (-5).
It seems the UVC requests are not handled correctly.
From the log, GET_DEF(PROBE) is requested but not answered correctly. Please check the VS request callback (demo_video_vs_request_process in previous example), maybe you can try to handle GET_DEF, like following:

```
    switch(wValue_CS)
    {
    case UX_DEVICE_CLASS_VIDEO_VS_PROBE_CONTROL:

        switch(bRequest)
        {
        case UX_DEVICE_CLASS_VIDEO_SET_CUR:

            /* Simply accept without change setting.  */
            status = UX_SUCCESS;
            break;

        case UX_DEVICE_CLASS_VIDEO_GET_DEF: // <- when GET_DEF requsted
        case UX_DEVICE_CLASS_VIDEO_GET_CUR:
        case UX_DEVICE_CLASS_VIDEO_GET_MIN:
        case UX_DEVICE_CLASS_VIDEO_GET_MAX:

            /* Simply send our data back (first 26 bytes).  */

```

Mahesh-dev-avula, 2023-03-08:
Hi @xiaocq2001 ,
Thank you so much for your inputs, 

Actually  VS request callback is not getting invoking because VS request callback get invoke only when   UX_SET_INTERFACE request comes from the host machine, but here my device not getting  UX_SET_INTERFACE request.

could you please explain that when device suppose to get  UX_SET_INTERFACE from host.

Regards
Mahesh Avula

xiaocq2001, 2023-03-08:
Usually UX_SET_INTERFACE is received when there is host application using the video camera, e.g., open a camera application that can preview the video input.

Mahesh-dev-avula, 2023-03-08:
@xiaocq2001 , Thank you for the quick replay.

Is host application for receive video stream data from device..?,

xiaocq2001, 2023-03-08:
Yes.
When host application starts receiving video stream data, the alternate setting is selected to indicate consuming USB bus bandwidth, then the ISO transfer starts to transfer video stream. When host application stops receiving video stream the interface alternate setting is back to 0 to reduce USB bus bandwidth consumed.

Mahesh-dev-avula, 2023-03-08:
@xiaocq2001 , got it  and thank you so much,

But still I have some doubt that as per the usbx stack When UX_SET_INTERFACE request comes from host then only Isochronous endpoint will be created and  VS request callback get invoked , demo_video_vs_request_process will set the probe control and commit control right..?

Once this process is completed then device will start send video streaming data to the host right..?

could you please correct above points.

Regards
Mahesh Avula

xiaocq2001, 2023-03-09:
Usually host set alternate setting to 0 and use VS request to probe control and commit control. Then host set interface to expected alternate setting and streaming data.

Mahesh-dev-avula, 2023-03-10:
Hi @xiaocq2001 ,

Got it and thank you so much.

Mahesh-dev-avula, 2023-03-10:
Hi @xiaocq2001 ,

host setting  alternate setting to 0 and  VS request callback getting invoking to probe control and commit control,

I can see these two requests UX_DEVICE_CLASS_VIDEO_GET_DEF and UX_DEVICE_CLASS_VIDEO_GET_CUR, 

UINT    demo_video_vs_request_process(UX_DEVICE_CLASS_VIDEO_STREAM *stream, UX_SLAVE_TRANSFER *transfer)
{

UINT                                    status = UX_ERROR;
UCHAR                                   bRequest;
USHORT                                  wValue_CS; /* In high byte.  */
USHORT                                  wLength;
UCHAR                                   *data;

    /* Decode setup packet.  */
    bRequest = transfer -> ux_slave_transfer_request_setup[UX_SETUP_REQUEST];
    wValue_CS = transfer -> ux_slave_transfer_request_setup[UX_SETUP_VALUE + 1];
    wLength = ux_utility_short_get(transfer -> ux_slave_transfer_request_setup + UX_SETUP_LENGTH);
    data = transfer -> ux_slave_transfer_request_data_pointer;

    /* Check CS.  */
    switch(wValue_CS)
    {
    case UX_DEVICE_CLASS_VIDEO_VS_PROBE_CONTROL:

        switch(bRequest)
        {
        case UX_DEVICE_CLASS_VIDEO_SET_CUR:

            /* Simply accept without change setting.  */
            status = UX_SUCCESS;
            break;

        case UX_DEVICE_CLASS_VIDEO_GET_CUR:
        case UX_DEVICE_CLASS_VIDEO_GET_MIN:
        case UX_DEVICE_CLASS_VIDEO_GET_MAX:

            /* Simply send our data back (first 26 bytes).  */

            /* Check request length.  */
            if (wLength < 26)
                break;

            /* Copy data for transfer.  */
            ux_utility_memory_copy(**data**, video_probe_commit_buffer, 26);
          **here, what is the  video_probe_commit_buffer ..? also we are not passing data pointer to any function below, could you please explain this**

            /* Transfer request.  */
            status = ux_device_stack_transfer_request(transfer, 26, wLength);
            break;

        default:
            break;
        }

        break;

    case UX_DEVICE_CLASS_VIDEO_VS_COMMIT_CONTROL:

        /* Simply accept request.  */
        status = UX_SUCCESS;
        break;

    /* CS not supported now.  */
    default:
        break;
    }

    return(status);
}

could you please help on this

Regards
Mahesh Avula

xiaocq2001, 2023-03-13:
This is my testing data for you to reference:
```
/* APP specific probe/commit settings.  */
static UCHAR                            video_probe_commit_buffer[48] = {
    W(0x0000), /* bmHint  */
    1, /* bFormatIndex  */
    1, /* bFrameIndex  */
    DW(UX_DEMO_FRAME_INTERVAL), /* dwFrameInterval  */
    W(0), /* wKeyFrameRate  */
    W(0), /* wPFrameRate  */
    W(0), /* wCompQuality  */
    W(0), /* wCompWindowSize  */
    W(0), /* wDelay  */
    DW(UX_DEMO_FRAME_DATA_SIZE), /* dwMaxVideoFrameSize  */
    DW(UX_DEMO_MAX_PAYLOAD_SIZE), /* dwMaxPayloadTransferSize  */
    DW(UX_DEMO_CLOCK_FREQUENCY), /* dwClockFrequency  */
    0x03, /* bmFramingInfo  */
    1, /* bPreferedVersion  */
    1, /* bMinVersion  */
    1, /* bMaxVersion  */
    0, /* bUsage  */
    0, /* bBitDepthLuma  */
    0x00, /* bmSettings  */
    0, /* bMaxNumberOfRefFramesPlus1  */
    W(0x0000), /* bmRateControlModes  */
    W(0), W(0), W(0), W(0) /* bmLayoutPerStream  */
};
```
For details about the structure and fields, please refer to USB Video spec.

Mahesh-dev-avula, 2023-03-13:
@xiaocq2001 ,
Thank you so much,
We need to send this structure info to the host when probe and commit request comes right..?
could you please share  usbx video class driver sample application, so that it will be easy to understand,

Mahesh-dev-avula, 2023-03-15:
Hi @xiaocq2001 ,
My device is enumerated at host and created node (/dev/video0)for it, I really thank you for your support.

One doubt that, from where ux_device_class_video_write_thread_entry get invoke, and could you please share any azure usbx documents for video stream data tranfer flow.

Regards
Mahesh Avula

xiaocq2001, 2023-03-17:
`ux_device_class_video_write_thread_entry` is passed to stack to start a thread to send data in FIFO to host, the FIFO is created from parameters passed in register function.
When you need to add data to FIFO, use `ux_device_class_video_write_payload_get` to get buffer to fill and use `ux_device_class_video_write_payload_commit` to commit to transfer.

BTW, which board are you working on for now?

Mahesh-dev-avula, 2023-03-17:
Hi @xiaocq2001,
Thanks for your support, I am using ASIC Board(with arm coretx m55 processor).
I was bit struggling to write video class application , earlier I have given support for CDC-ACM class support and that was working fine.

Regards
Mahesh Avula

Mahesh-dev-avula, 2023-03-20:
Hi @xiaocq2001 ,

Is there any open source host application for UVC driver to validate, or could you please suggest how the host application would be..?

Regards
Mahesh

xiaocq2001, 2023-03-23:
Here is some example (on ST NUCLEO-H723ZG) that you can reference:
https://github.com/STMicroelectronics/x-cube-azrtos-h7/tree/main/Projects/NUCLEO-H723ZG/Applications/USBX/Ux_Device_Video

Mahesh-dev-avula, 2023-03-23:
Hi @xiaocq2001 ,

Thank you so much for all your inputs

Mahesh-dev-avula, 2023-03-30:
Hi @xiaocq2001 ,

I am using host application as VLC media player to stream the data from the device, but device not receiving SET_INTERFACE request with the alternate setting is selected to 1.

I have observed one thing at host side while running VLC media player , i have collected dmesg logs that i am posting here.

**[20125207.047025] usb 1-4.4.4.2: new high-speed USB device number 65 using xhci_hcd
[20125207.147717] usb 1-4.4.4.2: New USB device found, idVendor=0525, idProduct=a4a7
[20125207.147719] usb 1-4.4.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[20125207.147720] usb 1-4.4.4.2: Product: Devkit
[20125207.147721] usb 1-4.4.4.2: Manufacturer: AlifSemiconductor
[20125207.147722] usb 1-4.4.4.2: SerialNumber: 1200
[20125207.272416] uvcvideo: Found UVC 1.10 device Devkit (0525:a4a7)
[20125207.644515] input: Devkit as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.4/1-4.4.4/1-4.4.4.2/1-4.4.4.2:1.0/input/input36
[20125207.650508] kauditd_printk_skb: 8 callbacks suppressed
[20125207.650511] audit: type=1400 audit(1680155724.219:25407): apparmor="DENIED" operation="open" profile="snap.notepad-plus-plus.notepad-plus-plus" name="/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.4/1-4.4.4/1-4.4.4.2/busnum" pid=18819 comm="winedevice.exe" requested_mask="r" denied_mask="r" fsuid=1002 ouid=0
[20125207.721901] audit: type=1400 audit(1680155724.291:25408): apparmor="DENIED" operation="open" profile="snap.notepad-plus-plus.notepad-plus-plus" name="/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.4/1-4.4.4/1-4.4.4.2/busnum" pid=18819 comm="winedevice.exe" requested_mask="r" denied_mask="r" fsuid=1002 ouid=0
[20125330.019469] uvcvideo: Failed to set UVC probe control : -110 (exp. 34).
[20125335.139257] uvcvideo: Failed to set UVC commit control : -110 (exp. 34).
[20125340.259294] uvcvideo: Failed to set UVC probe control : -110 (exp. 34).
[20125345.379047] uvcvideo: Failed to set UVC probe control : -110 (exp. 34).
[20125345.782295] uvcvideo: Failed to set UVC probe control : -71 (exp. 34).
[20125345.785626] uvcvideo: Failed to set UVC probe control : -71 (exp. 34).
[20125345.785684] uvcvideo: Failed to set UVC probe control : -71 (exp. 34).
root@alif:/home/mahesh#** 

Could you please help on this.

Regards
Mahesh Avula

xiaocq2001, 2023-03-30:
> [20125330.019469] uvcvideo: Failed to set UVC probe control : -110 (exp. 34).
> [20125335.139257] uvcvideo: Failed to set UVC commit control : -110 (exp. 34).
> [20125340.259294] uvcvideo: Failed to set [...]
I usually test video device under windows. But from upper log, it seems host send probe requests to device but fail. Maybe you can check if the VS control request callback in application is called and correct answer is reported to host.

Mahesh-dev-avula, 2023-04-05:
Hi @xiaocq2001 ,
Thank you for your inputs

**[20125330.019469] uvcvideo: Failed to set UVC probe control : -110 (exp. 34).
[20125335.139257] uvcvideo: Failed to set UVC commit control : -110 (exp. 34).
[20125340.259294] uvcvideo: Failed to set UVC probe control : -110 (exp. 34).
[20125345.379047] uvcvideo: Failed to set UVC probe control : -110 (exp. 34).
[20125345.782295] uvcvideo: Failed to set UVC probe control : -71 (exp. 34).
[20125345.785626] uvcvideo: Failed to set UVC probe control : -71 (exp. 34).
[20125345.785684] uvcvideo: Failed to set UVC probe control : -71 (exp. 34).**

These errors I have resolved, now i could see video probe control requests and commit control requests are comming also device also responding back to host.

 Host application as VLC media player sending UX_SET_INTERFACE request after(video probe control requests and commit control) with the alternate setting is selected to 0, but host app suppose to send UX_SET_INTERFACE request  with the alternate setting is selected to 1. 
I am attching screenshot for reference.

[image].

could you please help on this.

Regards
Mahesh Avula

Mahesh-dev-avula, 2023-04-06:
Hi @xiaocq2001 ,

One more thing I have observed that when we run host application as VLC media player , I could see below error that to not immediatly after running VLC media player after some time this error is comming.
[image]

could you please help on this.

Regards
Mahesh Avula

Mahesh-dev-avula, 2023-04-20:
Hi @xiaocq2001 ,

I am have tested usb video class driver on window machine and used host application as VLC media player, sending UX_SET_INTERFACE request after(video probe control requests and commit control) with the alternate setting is selected to 0, but host app suppose to send UX_SET_INTERFACE request with the alternate setting is selected to 1 to use USB bandwidth to stream data.
I would say that behaviour of my application is same on Windows machine as well as Linux machine, could you please help on this to move further, I am not understanding where it's going wrong.

Regards
Mahesh Avula

xiaocq2001, 2023-04-20:
Here is something under windows for you to reference:

The USB traces as a whole:
[image]

Get Cur/Max/Min content:
[image]

Mahesh-dev-avula, 2023-04-21:
@xiaocq2001 ,
Thank you so much for your inputs.

Mahesh-dev-avula, 2023-04-21:
Hi @xiaocq2001 ,

Yes, I have compared my usb analyzer trace with your screenshot , found very few differences, as part of video stream request i am not seeing usb commit control request.
I am attaching my screenshot here.
Starting USB trace when i tested on window machine.
[image]

[image]

One more difference i could see that configuration descriptor length as 174 in your case but my case configuration descriptor length as 181, will it make any difference?.

I really appriciate your help.

Regards
Mahesh Avula

Mahesh-dev-avula, 2023-04-21:
Hi @xiaocq2001 ,

When I tested on linux machine i could see that USB commit control request is comming also SET_INTERFACE request also comming from host,

I am attaching USB trace here.
[image]

[image]

I was thinking some where it's going wrong but i couldn't able to find that.

Regards
Mahesh Avula

Mahesh-dev-avula, 2023-04-24:
Hi @xiaocq2001 ,

Here I am attaching  usb video class analyzer  trace file, could you please check anything is wrong in the usb trace.
[USB_analyzer.zip](https://github.com/azure-rtos/usbx/files/11308669/USB_analyzer.zip)

if it's possible, could you please share your trace also.

Regards
Mahesh Avula

Mahesh-dev-avula, 2023-04-28:
@xiaocq2001 ,

Any inputs on my observations. Thanks

xiaocq2001, 2023-05-04:
From USB requests it seems there is no issue.
Maybe you can try more commonly seen video frame format for better host application support.

Mahesh-dev-avula, 2023-05-04:
@xiaocq2001 , Thank you for your inputs,
Could you please suggest which video frame format i should try, because first time i am trying on video class support.
I really appreciate your help.
Regards
Mahesh Avula

Mahesh-dev-avula, 2023-05-08:
@xiaocq2001 ,
If I am using different video frame format, what are the places i need to change in the descriptors, could you please give any example snippet for  different video frame format.

Regards
Mahesh Avula

xiaocq2001, 2023-05-09:
Please check "Universal Serial Bus Device Class Definition for Video Devices: Frame Based Payload" for details of the descriptor.
Maybe you can capture trace of existing USB Camera on market for reference, of just check the example trace in total phase data center.

Mahesh-dev-avula, 2023-05-29:
@xiaocq2001, Sure, thank you so much.

MaheshAvula-Alifsemi, 2025-05-20:
Hi @xiaocq2001 ,I am working on the UVC device side and have successfully implemented sending isochronous payloads — the streaming is working as expected. However, I noticed that the stream->ux_device_class_video_stream_buffer_error_count value is incrementing with every payload. I'm trying to understand whether we need to schedule payloads in advance. If so, could you please suggest how to properly implement this?