eclipse-threadx/usbx #89: Null pointer exception in the thread named "_ux_device_class_storage_thread"

HeBea, 2023-03-22:
**Describe the bug**
I run MSC class on my device successfully but with a very low probability of null pointer exceptions which lcated in _ux_device_class_storage_thread/_ux_device_class_storage_read/_ux_device_class_storage_csw_send/_ux_utility_memory_copy.  There is a loop in the thread, it runs all the time. I think that if ux_slave_transfer_request_data_pointer is set to NULL by CONFIGURA or otherwise，the loop will crash when do memory copy at this moment. It is just my thought. Could you please help on this inssue?  How to avoid null pointer exception?

HeBea, 2023-03-22:
Did anyone have the same experience?

xiaocq2001, 2023-03-24:
I did not see such exceptions on my side.

`ux_slave_transfer_request_data_pointer` is currently allocated in `ux_device_stack_initialize`, which is called before device class storage class registration and storage thread creation. It should be always available before device stack uninitialized.

HeBea, 2023-03-27:
Thank you for your reply.
There is a difference in my side. We free the memory of on-endpoint and set  ux_slave_transfer_request_data_pointer 
 to NULL in ux_slave_dcd_function when UX_DCD_DESTROY_ENDPOINT event comes. It may cause the loop to get a null pointer. I see that the demo code just set ux_sim_slave_ed_status to unused in the ux_slave_dcd_function. 
Should I do as the demo code? Do you have a better suggestion?

xiaocq2001, 2023-03-27:
Yes, please refer to demo code. Let stack manage the memory.

HeBea, 2023-03-28:
Shoul I set `ux_slave_transfer_request_data_pointer` to NULL when free the memory? I am not sure about this. If do not set it to NULL, the thread will write to the memory which is freed.

xiaocq2001, 2023-03-29:
The thread is created (in class initialization invoked by class registration) after buffer allocated (in stack initialization), and the buffer is not freed (in stack uninitialization) before thread deletion (in class uninitialization invoked by class unregistration).