eclipse-threadx/usbx #90: "_ux_device_class_dfu_entry" function does not support the "case UX_SLAVE_CLASS_COMMAND_UNINITIALIZE"

TakahiroMuranaka227, 2023-04-03:
### Issue
The "_ux_device_class_dfu_entry" function does not support the "case UX_SLAVE_CLASS_COMMAND_UNINITIALIZE". 
I think that it is necessary to support  "case UX_SLAVE_CLASS_COMMAND_UNINITIALIZE" in case as follow "_ux_device_class_dfu_entry" function.
 
======
 
UINT  _ux_device_class_dfu_entry(UX_SLAVE_CLASS_COMMAND *command)
{
    switch (command -> ux_slave_class_command_request)
    {
                  :
    case UX_SLAVE_CLASS_COMMAND_UNINITIALIZE:
 
        /* Call the init function of the CDC ACM class.  */
        status =  _ux_device_class_dfu_uninitialize(command);
        
        /* Return the completion status.  */
        return(status);
                  :
   }
}
 
UINT  _ux_device_class_dfu_uninitialize(UX_SLAVE_CLASS_COMMAND *command)
{
                                          
UX_SLAVE_CLASS_DFU          *dfu;
UX_SLAVE_CLASS              *class_ptr;
 
    /* Get the class container.  */
    class_ptr =  command -> ux_slave_class_command_class_ptr;
 
    /* Get the class instance in the container.  */
    dfu = (UX_SLAVE_CLASS_DFU *) class_ptr -> ux_slave_class_instance;
 
    /* Sanity check.  */
    if (dfu != UX_NULL)
    {
 
#if !defined(UX_DEVICE_STANDALONE)
 
        /* Free resources and return error.  */
        _ux_utility_thread_delete(&dfu -> ux_slave_class_dfu_thread);
        _ux_utility_event_flags_delete(&dfu -> ux_slave_class_dfu_event_flags_group);
        _ux_utility_memory_free(dfu -> ux_slave_class_dfu_thread_stack);
#endif
 
    }
            
    /* Return completion status.  */
    return(UX_SUCCESS);
}
 
======

xiaocq2001, 2023-06-28:
Thanks for the feedback. We will add it later.
BTW, memory for DFU instance is allocated in "initialization" and should be freed in "uninitialization".

ayedm1, 2025-04-27:
#159 fixed in this PR https://github.com/eclipse-threadx/usbx/pull/159
@fdesbiens please close this issue